Fluigent/fgt-SDK #3: fgt_set_sensorRegulationResponse not functional for vacuum-driven flow control.

tedstertm, 2019-05-21:
The sensor-based flow control defaults to a positive pressure based feedback. My instrument uses vacuum. As a result the fgt_set_sensorRegulationResponse function provides feedback inverse of what it needs to be to get to the appropriate flowrate.

CristianGlavana, 2019-05-22:
In most cases, sensor sense points out from the reservoir. This is the expected logic of the regulation algorithm. Thus there is an exception on vacuum MFCS™ Series instruments, sensor sense has to be inverted from nominal case: sensor has to point towards the reservoir.
MFCS instruments are always seen with positive range, enven if vacuum.

CristianGlavana, 2019-05-22:
In case you do not want to change the set-up, sensor can be tricked by applying a scale factor: fgt_set_sensorCustomScale(sensorIndex, -1, 0 ,0);

tedstertm, 2019-05-23:
I changed the scale as suggested but am no longer getting sensor readings in µL/min, but as some very long number. Is there a correction that I can apply elsewhere to correct this?

CristianGlavana, 2019-05-24:
An update was pushed on master branch. After investigation and testing in same conditions there seem to be no issue. After clonning could you please run the example Basic Sensor Regulation.cpp and send a screenshot of displayed terminal, this way we can deeper investigate if needed.

CristianGlavana, 2019-07-29:
Thank you for provided feedback and for participating on beta version. First release 19.0.0 is now available.
This issue will be closed but feel free to contact us for any question.

closed 2019-07-29 by CristianGlavana