Fluigent/fgt-SDK #4: Function to change regulation response time have no affect

VincentBressler, 2019-06-07:
We are finding slow regulation when using the API vs when using the application provided by Fluigent.

Specifically, calling this function with various responseTime values:
`Fgt_set_sensorRegulationResponse(unsigned int sensorIndex, unsigned int responseTime);`

//And calling this function with various "value" settings:
`Fgt_set_pressureResponse(unsigned int pressureIndex, unsigned char value);`

does not seem to have any affect on actual response time, which is very slow, particularly when we are within 10% of the target flow rate.

We are using the following API version:

```
/*============================================================================
*               Fluigent Software Developement Kit for C++
*----------------------------------------------------------------------------
*         Copyright (c) Fluigent 2019.  All Rights Reserved.
*----------------------------------------------------------------------------
*
* Title:   fgt_SDK_Cpp.cpp
* Purpose: Wrapper to fgt_SDK_*.dll
*			Contains an interface to each dll function and type conversions
* Version: 19.0.0.0
* Date:	06/2019
*============================================================================*/
```
Our target operating system is Windows x64.

Thanks for your help.
Vince

CristianGlavana, 2019-06-11:
Hi Vince,

Thank you for the feedback. There was indeed a small internal error concerning the regulation. 
New dll is released and pushed to master after test approval, same performances are met as in A-i-O software.

Concerning _Fgt_set_sensorRegulationResponse_ function responseTime minimal value is 2s(FlowEZ) or 6s(MFCS). Incresing this number slowers the regulation

Concerning Fgt_set_pressureResponse : for FlowEZ available values are 0: use of fast switch vales or 1: do not use fast switch vales. Default value is 0.
For MFCS available values are from 1 to 255. Higher the value, longer is the response time. Default value is 5.

Let me know if you have questions or other unexpected behaviour.

VincentBressler, 2019-06-11:
Thanks Christian.

Please give the URL for the latest "master", including the new DLL. Also, how will I know if the changes are included in the DLL?

CristianGlavana, 2019-06-12:
You can download again from master branch, same as for the first time. In order to check if all is ok you can run again your example, just change fgt_SDK_64.dll from Shared folder or run provided C++ example.

VincentBressler, 2019-06-14:
Is this the DLL?

https://github.com/Fluigent/fgt-SDK/blob/master/Shared/Windows/fgt_SDK_64.dll

This is not a valid DLL:
[image]

closed 2019-06-14 by VincentBressler

CristianGlavana, 2019-06-17:
Hi, did you managed to find error source? 
I also run Dependency Walker and dumpbin from VS tools without issue.
Dll depends only on kernel32.dll, it was built using Visual Studio 2019 which should have no impact.

VincentBressler, 2019-06-18:
I'm pretty sure that this is not anything that I am doing.
Please confirm that the URL above is correct. If it is, then there is something wrong with that DLL.
When I drop that DLL in to replace what I have been using, I get the above errors.

I have used GetLastError() to check the error code returned:

[image]

This is interesting, as I am running from a x64 Windows program and trying to load: fgt_SDK_64.dll

Here are all the fluigent DLLs in the directory:
[image]

Please confirm that you can load the DLL from a Windows x64 program.

CristianGlavana, 2019-06-18:
It seems that there is a download problem. fgt_SDK_64.dll file size is around 700 kB.
Please donwload it again using this URL: https://github.com/Fluigent/fgt-SDK/archive/master.zip
Or download it from the attached link:
[fgt-SDK-master.zip](https://github.com/Fluigent/fgt-SDK/files/3302398/fgt-SDK-master.zip)

VincentBressler, 2019-06-18:
Thanks. fgt_SDK_64.dll from the above link is loading. I'll let you know if this fixes the problem.

VincentBressler, 2019-07-03:
The new fgt_SDK_64.dll still has very slow regulation response, on the order of several minutes.

Is the source code for this DLL availaible?

CristianGlavana, 2019-07-09:
We can not reproduce such behaviour. Can you give more information about your microfluidic set-up and used pressure ranges and flow-units?
Can you try All-in-One software and check that response time is in your expected delay?
Can you please run provided example "Basic Sensor Regulation" and send console output?
We are sorry but as source code contains regulation algorithm we can not provide it.

CristianGlavana, 2019-07-18:
We detected an issue when multiple regulations are used.
You can download fixed version from master branch, it might resolve slow regulation behaviour.