Fluigent/fgt-SDK #5: Regulation with more than one flow meter fails

tedstertm, 2019-07-17:
While regulating pressure/flow using a single flow meter things work fine using the SDK. When both flow meters are called to perform regulation using independent pressure regulators regulation fails.

CristianGlavana, 2019-07-18:
Thank you for the feedback and we are sorry for this desagrement.
Full release is planned next week, beta tests are still going on in order to ensure that there is no major issue. 
Thank you for your contribution, indeed when multiple regulations a bad allocation used same instance insted of separate ones.

You can download the dll fix from master branch.

Please feel free if any question on this SDK and regulation.

CristianGlavana, 2019-07-29:
Thank you for provided feedback and for participating on beta version. First release 19.0.0 is now available.
This issue will be closed but feel free to contact us for any question.

closed 2019-07-29 by CristianGlavana