Fluigent/fgt-SDK #7: calling fgt_set_pressure causes unhandelable exception

VincentBressler, 2019-07-19:
This is intermittent behavior. Adding a delay before calling the function does not fix the problem. 

I am using DLLs that I downloaded from here today:
https://github.com/Fluigent/fgt-SDK/archive/master.zip

CristianGlavana, 2019-07-19:
Did you have this issue before? Changes on the library did not cencern fgt_set_pressure function.
We are proceeding final tests for a release next week. Sending over 3million consecutive orders, without delay caused no problem. Dll also supports multithreading. Does this happen with the basic example?
If you have additional information it is welcomed. For now we may need more time to investigate for such issue.
Thank you

VincentBressler, 2019-07-19:
I believe that this behavior is new. I'll do some testing to confirm.

VincentBressler, 2019-07-19:
The problem has been around for at least a month. Here is the sequence of calls which intermittently cause the problem:
[image]

VincentBressler, 2019-07-19:
I have absolutely confirmed that the exception occurs inside a call to pfgt_set_pressure.
I have put exception handling round this function call using both

__try  and try

Exception handling does not work in either case. It's almost as if an ExitProcess() call is happening somewhere in your code.

VincentBressler, 2019-07-19:
Improved logging, included the pressure value being set:
[image]

VincentBressler, 2019-07-19:
Adding long delays (10 seconds or more) after set_sensorRegulation () does not make any difference.
Adding long delays (1 second) before each call to set_pressure() does not make any difference.

I got rid of these delays because they simply slow down debugging.

CristianGlavana, 2019-07-23:
Thank you for the much provided details.
Indeed an exception is randomly thrown when regulation is running and pressure orders are send. A concurrent ressource was used in this case.
Libraries are recompiled with apropriate synchronisation. It is now available on master branch.

Please feel free to contact us if any question.
Thank you for the detailed feed back and for your colaboration!

VincentBressler, 2019-07-23:
Thanks! I have confirmed that the new version of: 
fgt_SDK_64.dll 
fixes the problem.

CristianGlavana, 2019-07-29:
Thank you for provided feedback and for participating on beta version. First release 19.0.0 is now available.
This issue will be closed but feel free to contact us for any question.

closed 2019-07-29 by CristianGlavana