Fluigent/fgt-SDK #13: unable to detect a controller in sdk, but can use Maesflo

stevhud5, 2019-09-03:
Dear Fluigent,
I cannot detect a controller using two Windows 10 machines, but I can use Maseflo 3.3.1. (I haven't tried Maesflo on both of them.) 
I can detect a controller using the sdk on Windows 7 machines. 
- On both types of machines I am using an Anaconda distribution of Python 3.6 or 3.7. Spyder is our IDE.  
- The controller is an MFCS-8C 1000mbar, whose sn is 0106.
- Mark Aruda sent me a report assistant tool, but that did not run properly on my machine. 
Thanks for your help. 
Steve Hudson

CristianGlavana, 2019-09-04:
Hello,

Some old MFCS devices might not be supported by our latest softwares and SDK. 
If this is the case please use attached SDK, during initialization step set 0 as function parameter.

[Fluigent_SDK_old.zip](https://github.com/Fluigent/fgt-SDK/files/3575464/Fluigent_SDK_old.zip)

If you successfully initialized the MFCS on another computer maybe a dependency is missing. In this case you can try to install VC redistrubution from Microsoft: https://support.microsoft.com/en-us/help/2977003/the-latest-supported-visual-c-downloads

Python does not support dll calls natively, it is dependent on microsoft services.

Please let me know about the advancement of this issue

Best regards

stevhud5, 2019-09-04:
Thanks Cristian.
I am trying to use this. 
I found one bug in Python_MFCS_Series.py
in line 44, mfcsInit = lib_mfcs.mfcs_initialization
should be respelled to: mfcsInit = lib_mfcs.mfcs_initialisation

I am getting error = 1, from mfcs_get_serial

Thanks.
Steve

stevhud5, 2019-09-05:
Cristian, I have a question about my previous comment. I don't know what's in each dll, but I was working off of the contents of .h files. The .h file under C++ and the corresponding one under Python are different. The behavior I observe (in python) seems consistent with the one under C++. 
Specifically, I first got an error that function mfcs_initialization was not found. I changed it to mfcs_initialisation and it found it. The C++ .h has the spelling with s and the python .h has the spelling with z. Although it found mfcs_initialisation, it does not find mfcs_set_pressure. mfcs_set_pressure is listed in the python .h, but not in the C++ one. 
Does any of this information give a clue about the problem?
Thanks for your help.

CristianGlavana, 2019-09-05:
Sorry about that, I sent you an intermediate version. Indeed in order to initialize communication with the device "mfcs_initialisation(0)" function has to be called. Also in order to set pressure use "mfcs_set_auto" function.
Here is associated documentation: 
[MFCS Series SDK Manual.pdf](https://github.com/Fluigent/fgt-SDK/files/3579677/MFCS.Series.SDK.Manual.pdf)

If you get a return error of 1 when calling functions is mainly that the device is not connected (powered on and pluged to the coputer).

stevhud5, 2019-09-06:
Here is a little more info from running the code:
MFCS says it is initialized and returns a handle. 
It fails to get SN with error code =1
The program proceeds. After setting, the pressure reading remains at 0 mbar.
The code finishes apparently normally. 
But this happens with both Win10 and Win7 machines, and I cannot therefore operate the mfcs. 

I have another dll (which we have used previously, before my current inquiry.) This works well on Win 7 but not on Win10. The issue here I think is addressed in the thread at https://stackoverflow.com/questions/53182796/python-ctypes-issue-on-different-oses 
Please advise. 
Thanks.

CristianGlavana, 2019-09-06:
We are sorry for about encountered issue. If program executes without exception is thrown, calling MFCS dll seems not to be the issue. Returned code of 1 when trying to get the SN is more a problem, Communication with the device is not established although a handle value is always returned.

Can you please add, if not the case, a small delay after initialization step: "time.sleep(1)"

We encountered same ctypes issue as the stackoverflow thread. Initialization function is using restype and argtypes; after testing other functions do not require explicit declaration.

As this issue might be difficult to find (computer configuration, hardware...) a TeamViewer session might be more effective for troubleshooting if ok for you?

stevhud5, 2019-09-06:
Thanks Cristian.
yes, I think it would be good to arrange a TeamViewer session, which I
think I can do with my (windows 10) tablet when I am off campus.
Steve

stevhud5, 2019-09-09:
Dear Cristian,
Could we connect Monday? I will install teamviewer. Can I install for
one-time use?
Please let me know if there is suitable time for you.  Thank you very much.
Steve

stevhud5, 2019-09-09:
Dear Cristian,
Do you have time to connect by teamviewer today?
Please let me know. Thanks.

CristianGlavana, 2019-09-09:
Hello,
I sent you an email for more confidentiality. Please tell me if you have Teamviewer link.
I have one hour left today if ok for you or tomorrow starting from 2PM Paris time.

stevhud5, 2019-09-09:
Oh, Thank you for your reply. I am sorry that I did not see it right away.
I will install team viewer for single use.
Steve

stevhud5, 2019-09-09:
Hi Cristian,

My id is 1 384 772 521
Thanks,
Steve

stevhud5, 2019-09-09:
Cristian,
I am set up here.

stevhud5, 2019-09-09:
Hi Cristian,
If you prefer, we can connect tomorrow at about 2 pm your time.
Thanks.
If you still want to try today, I have it all set with team viewer id 1 384
772 521
Steve

stevhud5, 2019-09-09:
Also, Cristian, please let me know if you prefer for me to email you at
another address.
Thanks.
Steve

CristianGlavana, 2019-09-09:
Hi,
I am sorry, just seeing your messages. 
Can we schedule this session for tomorrow, today it might be too short if further invertigation is required. This way we may also take some time to discuss your application if needed.
TYhank you,
Cristian

MathieuSteiner, 2019-09-10:
> Hi Cristian, My id is - removed by admin - this is a public thread and we prefer to not have such private information shown Thanks, Steve

Hi @stevhud5 
This is a public thread. We consider that it is not a good habit to share your computer ID in public
We have thus removed any mention to your computer ID for now

Cristian will get in touch with you

closed 2019-09-10 by CristianGlavana

micropolimi, 2019-10-18:
> Dear Fluigent,
> I cannot detect a controller using two Windows 10 machines, but I can use Maseflo 3.3.1. (I haven't tried Maesflo on both of them.)
> I can detect a controller using the sdk on Windows [...]

Hi @CristianGlavana, I have the same problem of @stevhud5 . I have already tried to do what you suggested but still I fail in getting the MFCS SN.  Could you please tell me how did you solve the problem?

Thanks :)