Fluigent/fgt-SDK #14: SDK installation issue

SDKinstallationissue, 2019-09-09:
Dear Sir/Madam,
I would like to have modules for setting pressures onto "MFCS-EZ" device
and monitoring flow rates with "flow units".
Can you tell me how to achieve this ?
I thank you in advance for your help,
and I wish you a nice day,
Christian

CristianGlavana, 2019-09-09:
Hello,
You can download latest release using this link: [https://github.com/Fluigent/fgt-SDK/releases/latest](Fgt-SDK-release)

If you are using one of five directly supported languages (C++, C#, Python, MATLAB or LabVIEW) you will find an interface with examples and specific documentation. You can also directly use "fgt_SDK_*.dll" file.

Please let me know if any issue.

CristianGlavana, 2019-09-09:
Sorry bad link, here is the correct: [Fgt-SDK-release](https://github.com/Fluigent/fgt-SDK/releases/latest)

SDKinstallationissue, 2019-09-09:
Hi Mr. Glavana,
Thank you for the rapidity of your answer.
Using this link I could download a Labview project.
Unfortunately it doesn't work for being unable to find some libraries.
Could it be that it requires some specific installation ?

CristianGlavana, 2019-09-09:
In the archieve (fgt-SDK.rar) LabVIEW folder contains "fluigent_lib_fluigent_sdk-19.0.0.5.vip" file.
Use VI Package Manager (provided by JKI) in order to install it in your LabVIEW developement environement. Package contains a palette, documentation and several examples.

SDKinstallationissue, 2019-09-09:
I installed the package.
Now I can launch the examples but they all display the error "Module_is_sleep" has occurred ...

SDKinstallationissue, 2019-09-09:
or "No_instr_found"

CristianGlavana, 2019-09-09:
This error is thrown when the instrument is powered on but not running. Press the "play" button on the MFCS-EZ, led light should swith to green.
For "No_instr_found" error please check that instruments are plugged in (USB cable) and powered on.

SDKinstallationissue, 2019-09-09:
I am confused: the MFCS was on but not started.
Now I will try to get the modules and the devices to communicate together.
I wish you a nice afternoon.

SDKinstallationissue, 2019-09-09:
Would you have just the basic modules ?
- find MFCS
- find flow units
- set pressures
- read flow rates

CristianGlavana, 2019-09-09:
On LabVIEW under Help->Find Examples.. search for "Fluigent SDK".
"Basic Get Instruments Info" will list all found instruments and channels.
You cas see "Basic Set Pressure", "Basic Set Regulation" for pressure and flowrate control and read.

Otherwise "fgt set pressure" and "fgt get sensor value" functions can be found on "Fluigent>FluigentSDK" palette (right click on diagram) but we recommend first starting with the examples.

SDKinstallationissue, 2019-09-09:
Thank you.
I will try to manage with those.

SDKinstallationissue, 2019-09-09:
I have one more problem:
the software detects the flow rate platform but not the 4 flow units that are plugged into it.
Would you have an advice for this also ?

MathieuSteiner, 2019-09-10:
Hi

1. Please connect the Flowboard and Flow units before opening LabVIEW
2. Open LabVIEW
3. Run the "Basic Get Instrument Info.vi" example
4. Take a screenshot of the second tab (a.k.a "Channels information") (so that we can be sure that flow units are indeed not detected)

Thanks

SDKinstallationissue, 2019-09-10:
Hi
Since It was not working I thought that maybe the package was not properly installed.
So I uninstalled it to run the installation again.
And now it is been 10 times that the installation timed out.
I will let you know as soon as it will be possible again.
Thank you for your help

MathieuSteiner, 2019-09-10:
Hi

It is not normal that installing the package takes so much time
You can read the following KB if you experience difficulties using VIPM: http://www.ni.com/tutorial/54770/en/

Once installed, this is how your package should look like (in my case, I have it installed for LabVIEW 2017 64 bits)
[image]

SDKinstallationissue, 2019-09-10:
The package manager fails at connecting to Labview.
I do not know why. Yesterday it succeeded at the first try.

SDKinstallationissue, 2019-09-10:
Finally it worked.
Here is the screenshot.
[image]

MathieuSteiner, 2019-09-10:
> Finally it worked.
> Here is the screenshot.
> [image]

Hi

Please:

1. Close LabVIEW
2. Disconnect your MFCS from the computer
3. Connect only your Flow board with all Flow units connected
4. Launch LabVIEW
5. Run the same example again
6. Provide screenshots of both tabs

Thanks

SDKinstallationissue, 2019-09-10:
Here they are:
[image]
[image]

SDKinstallationissue, 2019-09-10:
Is it what you expected ?

MathieuSteiner, 2019-09-10:
> Is it what you expected ?

Yes. 

1. Please confirm that your Flowboard power LED is green and lighted when it's connected to the computer.
2. Please get USBDeview (https://usbdeview.en.uptodown.com/windows), run it and make sure that there is a Flowboard in green

[image]

As an example, here is what the first tab should display when there is a Flowboard detected
[image]

SDKinstallationissue, 2019-09-10:
Yes: the green power led is on

[image]

MathieuSteiner, 2019-09-10:
It looks like your Flowboard is seen by your computer.

I would suggest removing all Flow units and run the LabVIEW example again. 
If the first tab displays a controller, then, it means that some Flow units is causing an issue

If the first tab remains empty, we will have to run a tool colled FRAT (Fluigent Report Assistant Tool), or A-i-O / MAT to check whether or not the hardware is detected by any of these software tools. Do you have any of these tools (they come along with our Fluigent Software Platform usb key)

SDKinstallationissue, 2019-09-10:
Here is what it gives:
[image]

SDKinstallationissue, 2019-09-10:
Unfortunately I have to leave. And I will be out of town tomorrow.
So I will try to contact you again on thursday.
I thank you for your help, and I wish you a nice evening,
Christian

MathieuSteiner, 2019-09-10:
> Here is what it gives:
> [image]

The Flowboard has appeared. Did you disconnect all Flow units to have it appear ?

MathieuSteiner, 2019-09-13:
From our TeamViewer sessions, the InitEx call is the source of the problem
We suspect that the problem lies in the conversion of the Flowboard SN. We will investigate and try to fix the issue.

closed 2019-09-13 by MathieuSteiner