Fluigent/fgt-SDK #16: Engine does not see FUs when using InitEx

MathieuSteiner, 2019-09-13:
The behavior is still to be confirmed (to be tested with a FB S/N > 9)
Using:

- LabVIEW 2019 and SDK 19.0.0.5
- a FB (S/N 359) with 4 FUs
- the example "Basic Get Instruments Info.vi"

No sensor channels is detected:
- fgt_get_sensorChannelCount return a count of 0
- fgt_get_sensorChannelsInfo returns an error 11 (both are coherent)

Removing the call to fgt_initEx solves the FUs detection and all 4 Flow units are now properly seen

MathieuSteiner, 2019-09-13:
Reproduced with FB S/N 1037
Replacing the InitEx by the Init method results in FUs being detected

CristianGlavana, 2019-09-20:
Released version 19.0.0.6
https://github.com/Fluigent/fgt-SDK/releases/tag/19.0.0.6

closed 2019-09-20 by CristianGlavana