Fluigent/fgt-SDK #19: LabView executable VI issues

lucasfrenz, 2019-10-02:
Hi,

I'm experiencing issues when trying to create an executable LabView VI.

The simplest way to reproduce this is to create a LabView project and pull the SDK example VIs into the project. Then create an .exe executable under build.
It won't control the pump and only show an error code (see attached screenshot). On the other hand running the VI regularly within Labview works fine.

The method of creating an executable with the Fluigent SDK worked fine for an older release I tried (1.0.0.13).

I'm using Windows 10 and LabView 2018 32bit.

[image]

Ygor-Oliveira, 2019-10-02:
Hello,

This is due to the fact that there is an indirect dependency on some of the shared libraries, which LabVIEW does not detect, so they are not copied to the build directory. We will try to fix it for the next release. In the mean time, there is an easy workaround:

**1.** Open your project dependencies and locate the fgt_SDK_32.dll item. Right-click on it and select Explore to open its containing folder.

[image]

**2.** Select the other 32-bit DLLs in the folder and Drag/drop them in your project. They can be added at the top level or in a virtual folder.

[image]
[image]

**3.** Open your build configuration and, under the Source Files section, select the DLLs you added to your project and add them to Always Included.

[image]

Now, when you build your project, it should copy the necessary files to the data directory of your build output and run correctly.

Let us know if this method works for you.

Best regards,
Ygor

lucasfrenz, 2019-10-02:
Worked perfectly. Thanks for the quick solution !

closed 2019-10-03 by gumen38