Fluigent/fgt-SDK #20: Unable to detect a MFCS DEVICEin sdk

matteomarsano, 2019-10-21:


CristianGlavana, 2019-10-21:
Hello,
This behaviour concerns few instruments, if your instrument serial number is lower than 255.
In this case calling "fgt_initEx" function has to be used. Parameter (serial number) should be 65,536‬ - MFCS SN (displayed on label).
You can call "fgt_detect" in order to check that instrument is properly detected and send us a feedback.

matteomarsano, 2019-10-22:
I tried to use 65,536 and "65,536" as input of fgt_init(). And in both the cases I get an error in fgt_initEx. By the way my device is a MFCS-FLEX 4C P/N:11000004 S/N:373, I don't know if it's in the category you mentioned.

CristianGlavana, 2019-10-22:
We might need to investigate this issue.
What is the exact error number that are you getting? Can you please call "fgt_detect()" and send prompt output?
If nothing is returned by detection fuction can you please use a debugger tool such as debugview:
https://docs.microsoft.com/en-us/sysinternals/downloads/debugview in order to catch internal SDK report and send it for analysis?

matteomarsano, 2019-10-22:
If I run the code BasicGetInstrumentInfo.py what I have is:
**WARNING:root:fgt_init error 11 - No instr found**

Here you can find the image of the prompt output.
Image: https://user-images.githubusercontent.com/56343263/67283972-99105180-f4cc-11e9-9533-3d48d2fb3fe5.jpeg

CristianGlavana, 2019-10-22:
For further investigation can you please also send prompt capture when calling "fgt_detect" function?
Also if you do not mind a capture using debugview tool. As SDK library is compiled it is only way to know what is going wrong inside.

matteomarsano, 2019-10-24:
When the function "fgt_detect" is colled we obtain this result in the debugview: 
[debugview.LOG](https://github.com/Fluigent/fgt-SDK/files/3766472/debugview.LOG).
What do you mean talking about prompt capture?

CristianGlavana, 2019-10-25:
Hello,
After investigation on similar hardware set-up there is indeed an issue when trying to detect the instrument.
This SDK does not support hardware manufactured before march 2013.
Older version of MFCS SDK can be used instead, here are the libraries and a python example: 
[MFCS_python_old.zip](https://github.com/Fluigent/fgt-SDK/files/3772322/MFCS_python_old.zip)
Other solution is to upgrade your MFCS hardware by sending it to us. If interested you can ask us for a quote at support@fluigent.com

closed 2019-11-21 by CristianGlavana