Fluigent/fgt-SDK #21: Fluigent | Matalb Control Code | Clemson University

omkarshukla10, 2019-12-02:
Dear Mark and Team members, 

I am facing an issue with our Matlab control Code for the system we have at Clemson University.  The code written by us runs fine but suddenly it stops working in the middle at any instance of time. We have over 400 mixing test samples and the code stops randomly at 6th case or 1st or 2nd during our testing. I am worried about this and I would like to seek help from your software engineer regarding the issue. 

I am attaching our code along with this email for your reference. I am also attaching the error messages. Would you able to help us in solving this problem? 

[image]
[image]

Let me know if your engineer requires a remote access to our system, we will be happy to provide the remote control of our system. 

Wishing you a Happy Thanksgiving, 
Omkar

[Matlab_code_omkar.txt](https://github.com/Fluigent/fgt-SDK/files/3911777/Matlab_code_omkar.txt)

CristianGlavana, 2019-12-02:
Hello Omkar,

We are sorry that you are not having a smooth experience using your M-Switch valve.
Actually we are working on a software issue which can cause valve to block randomly, especially when many requests occur in little time. One inconvenient work around today is to remove the valve from the switchboard and plug it again.
We will keep you informed about the fix.

Best regards,
Cristian

CristianGlavana, 2020-03-03:
New valve firmware fix this issue

closed 2020-03-03 by CristianGlavana