Fluigent/fgt-SDK #22: Get FlowEZ status - Labview

Cassaday, 2019-12-11:
i'm using the LabVIEW Lineup SDK
Get FlowEZ status , if a flow sensor is attached for the Boolean array element 'flow unit' this should return true ?  in testing I have a LINK-FlowEZ with a flow sensor attached and I read back a false value, but conversely when calling  GetFUType it does return the appropriate type of flow sensor attached.
i'm using an index array in expanded mode to get all the values and populate a cluster with all relevant information regarding the particular FlowEZ module that i'm accessing thru the LINK.

gumen38, 2019-12-12:
Hi

Could you please tell us what kind of Flow EZ you have
And post some code for us to be able to check it and try reproducing

Thanks

Cassaday, 2019-12-13:
[TestPanelFluigent.zip](https://github.com/Fluigent/fgt-SDK/files/3962459/TestPanelFluigent.zip)
just starting to write some code to integrate a fluigent driver into our automation software.  the Boolean array last element - is this a true false of when a flow sensor is attached ? when I have one attached it returns false but the 'GetFuType' returns with the type of flow sensor I have attached. I tested multiple times with varying scenarios of 1-4 flow EZ in series, they have been 7000mbar and 345mbar units, and I had M and L series alternating conditions of random units had sensors attached, etc.  everything seemed to be spot on except the value of that Boolean, thank you !

gumen38, 2019-12-16:
> [TestPanelFluigent.zip](https://github.com/Fluigent/fgt-SDK/files/3962459/TestPanelFluigent.zip)
> just starting to write some code to integrate a fluigent driver into our automation software. the [...]

Hi

Could you please downgrade your LV code to LV 2017 ?
I cannot open the file you provided and I would like to check a few things

Thanks

Cassaday, 2019-12-17:
SDK version = 1.0.0.13
i'm trying to save for prevision version - i'm getting a password protect on the VIs used from the SDK, can you provide me what it is ? thanks !

gumen38, 2019-12-17:
Thanks for this last information. A snipet or screenshot is enough

Cassaday, 2019-12-17:
[image]
[image]

gumen38, 2019-12-18:
Thanks for the screenshot. The Flow unit presence is on bit 7, not a bit 6

Bit 6 is somehow not carrying any information.
[image]

Just extend your array indexing primitive and wire the most significant bit (bit 7) to your Flow unit cluster field rather than bit 6 and you should be fine
[image]

Cassaday, 2019-12-18:
ok thank you for pointing this out, code changed and I've verified its tracks sensor there and not there, thank you !!
I started running some set pressure and set flowrate activity, 
Question : if I set flowrate, does the unit then work to maintain this flowrate, or is their an additional flag to set to make it operate this way ?  
Question : Set Pressure = 0 , is this the correct way to stop flow ? 
Thank you !!

Cassaday, 2019-12-18:
I also wanted to confirm, the tracking of the bits I do have wired to the indicators, are those correct ?
thank you !
Jason

FINCMAR, 2019-12-19:
Hi Jason, 

Its Mark, for the stop flow setting the pressure to zero is one way of doing it. I am not exactly sure of what your setup is at the moment, but you could also set one of the valves to go to a closed/plugged position. 

We typically recommend using a valve to stop the flow as when you set the pressure to zero it may take some time for the flow to go to zero depending on how resistive your fluidic channels are.  

I will let @gumen38 answer the other questions. 

Let me know if this is clear or not, feel free to call as well! 

Best, 
Mark

gumen38, 2019-12-19:
> ok thank you for pointing this out, code changed and I've verified its tracks sensor there and not there, thank you !!
> I started running some set pressure and set flowrate activity,
> Question : if I [...]

Set Flowrate is an order sent to the instrument. It then takes care of regulation; i.e. you do not have code any regulation algorithm in your own code.

As soon as you send a pressure order using the Set pressure VI, the flowrate regulation will be stopped, and the instrument will enter pressure regulation mode

gumen38, 2019-12-19:
> m, the tracking of the bits I do have wired to the ind

Other bits look fine to me

closed 2019-12-19 by gumen38