Fluigent/fgt-SDK #23: BasicSetPressure.py not working

jacks100, 2020-06-30:
I am trying to run BasicSetPressure.py using the python sdk. I think that I have successfully installed the SDK. I am trying to control a Flow-EZ device which is connected to a Link-EZ device. The Link-EZ device is connected to my computer. Python successfully executes the import statements at the top of the file. The program stops running when it attempts to execute "fgt_init()". It does not crash with an error message. Instead, it appears to almost freeze. The same issue occurs if I comment out the "fgt_init()" and move straight to "fgt_set_pressure(0,20)". 

I am using Python 3.8 on a 64-bit computer running Windows 10 Home.

Do you have any idea what may be causing this issue?

Ygor-Oliveira, 2020-07-01:
Hello, 

To confirm that the SDK is installed correctly, could you try running the unmodified example without any instruments connected to the computer? You should get the following output: 

```
WARNING:root:fgt_init error 11 - No instr found
WARNING:root:fgt_set_pressure pressure channel 0 of type NONE error 11 - No instrument found
Waiting 5 seconds...
WARNING:root:fgt_get_pressure pressure channel 0 of type NONE error 11 - No instrument found
Current pressure: nan
WARNING:root:fgt_get_pressureRange pressure channel 0 of type NONE error 11 - No instrument found
WARNING:root:fgt_set_pressure pressure channel 0 of type NONE error 11 - No instrument found
```

With the Link and Flow EZ connected to the computer, does the script get stuck indefinitely and has to be stopped by pressing Ctrl+C? The instrument might have a configuration problem that is causing it to send unexpected messages to the SDK. Could you try to launch the All-in-One application with the same instrument to see if it works correctly? The application is available to download at https://www.fluigent.com/download/

Let us know the results so we can pinpoint the cause of your problem.

Ygor

CristianGlavana, 2020-07-01:
Hello,

It seems that there is an issue with the detection or the initialization of the instrument. 
In adition with my coleague advice can you provide us with aditional debugging traces? They are generated by the SDK and can be catched using a generic Microsoft tool such as dbgview from the following link: https://docs.microsoft.com/en-us/sysinternals/downloads/debugview
After launching this software with admin rights check “Capture Win32”, “Capture Win32 Globals”, “Capture Kernel” and “Capture Events” from Capture menu.
Run your python script and export generated log.

mtrellet, 2021-05-21:
@jacks100 Hello, do you think we can close this issue?