Fluigent/fgt-SDK #24: LabVIEW wrapper 19.0.0.6 - fgt_SENSOR_Type - Missing L Dual in typeDef

gumen38, 2020-09-03:
While working on a test bench, I realized that my FLOW UNIT L dual was detected as a FLOW UNIT XL
This is due to the fact that the typeDefed enumerated does not contain value 7

[image]

CristianGlavana, 2020-09-03:
Flow unit L is implemented for both single and dual types. Maybe LabVIEW enum is shifted as value 7 corresponds to L_dual and 8 to XL_single. in the dll API

gumen38, 2020-09-04:
> Flow unit L is implemented for both single and dual types. Maybe LabVIEW enum is shifted as value 7 corresponds to L_dual and 8 to XL_single. in the dll API

The issue lies in the typeDef that misses one value. Not in the SDK core

CristianGlavana, 2020-12-23:
Fixed version 20.0.0.0 #26

closed 2020-12-23 by CristianGlavana