Fluigent/fgt-SDK #25: Set flow rate block in MAT not responding/working

Luukvdschepop, 2020-09-10:
Hi,
I am having troubles with the automation tool (MAT). I can automate pressures using MFCS --> Channel x --> set pressure at x mbar and then connect this block to a wait block of x minutes so that the pressure stays consant.

However, when I want to automate the flow instead of the pressure, it does not seem to work. I tried making a block using FRP --> flowboard --> flow unit x --> set flow rate --> set flow rate at x µL/min with pressure source x and connect this to a waiting block. This time around, the flow remains unchanged.

When I tried by tuning the pressure, the flow meter sensed flow. I am using an flow meter (size M), a flow rate platform, and an MFCS EZ. I can also send pictures if that is more helpful. Thanks in advance for the help!
Kind regards,
Luuk
P.S. I am new to GitHub so I hope this is the right spot to ask such questions ;)

gumen38, 2020-09-10:
Hi Luuk

This morning, we received your request by email. I was told that it was about LabVIEW SDK; this is why I suggested you to come and join here, so that we could discuss directly.

It seems that your problem is with MAT software right ?
The question does not really belong here in such case.

Let's try answering it nevertheless.

What version of MAT are you using ?
What range is your MFCS channel ?
How big is your reservoir ?

Thanks in advance for all these answers

Luukvdschepop, 2020-09-11:
Thanks for the response,
I am currently using MAT 19.0.0.1
I am using the MFCS EZ 0-2000 mbar (EZ-01000002)
The reservoirs are 2 mL big (eppendorf tubes)
Thanks for the help!
Luuk

gumen38, 2020-09-11:
Hi Luuk

I have tried with a 1 bar MFCS + M FU
With MAT 19.0.0.1. I am able to set the flowrate to the desired value (let's given my test setup I went to 3ul/min, and also tested 2 and 1 ul/min)

I cannot tell you why it does not work on your end. I'll forward your case to the support team and they will take care of your issue

Thanks

mtrellet, 2021-05-21:
@Luukvdschepop Hello, do you think that we can close this issue?