Fluigent/fgt-SDK #29: No instrument found when using LabView

jiaxinp, 2021-05-03:
I tried using Labview to read data from my new IPS sensors. When I run the Fluigent All-in-One software, the two sensors can be detected without any issues. However, when I try to connect to them using Labview, i get the error "No instrument found"  when I run the "Basic Read Sensor VI" and no sensors detected when I run "Basic Get Instruments Info" . What is the installation procedure for LabView? Is there something else I need to install? I looked through an old thread but couldn't find a solution (delete fgtintex?).

mtrellet, 2021-05-03:
Hello,
Thanks for contacting us. Would you mind providing the following information:
- Serial numbers of your IPS sensors
- Version of A-i-O that detected these sensors

We assume that you are using the last version of the SDK (`20.0.0`)?

If you prefer, you can provide those information by email: mikael.trellet@fluigent.com

Thanks in advance

jiaxinp, 2021-05-03:
Hi Mikael,

Thanks for your prompt reply. Here is the requested information:

   - Serial numbers of your IPS sensors: ips sn 10510 and ips sn 10511
   - Version of A-i-O that detected these sensor: Fluigent All-in-One 2020
   v20.0

Yes I downloaded the latest VIs from the fgtsdk repo.

Hope to hear from you soon.

Regards,
Jess

closed 2021-05-08 by jiaxinp