Fluigent/fgt-SDK #31: MFCS-EZ SDK issue: Instrument not found

medinahamidovic, 2021-05-19:
Hello!
I have been trying to set up a connection to control our MFCS-EZ either through Matlab or Python. I have installed SDK for both Matlab and Python, and no matter what I try fgt_init always returns an error: Instrument not found
I have tried on several computers just to be sure that my PC is not the problem.
Pump is working normal when using your softwares (AiO or MAT).

What could be the issue?

Medina

mtrellet, 2021-05-19:
Hello,

Sorry to hear that. 
Could you provide us the serial number of your MFCS (can be found in A-i-O for instance)? And the SDK version you are using?

Does the SDK work with other Fluigent instruments?

Thanks in advance for the information,

Mikael

medinahamidovic, 2021-05-20:
Dear Mikael,

the serial number is SN 1541
We have the latest SDK version downloaded 21.0.0
We do not have any other Fluigent instruments, just this controller.

Medina

medinahamidovic, 2021-05-20:
Dear Mikael,

we have now found the problem. The controller responds when we use the Legacy library but it is not responding when we use fgt_SDK.
How could we fix this so that it also works with fgt_SDK?

mtrellet, 2021-05-20:
Hello Medina,

Thanks for the information. We are currently investigating the issue. It seems that some MFCS-EZ do have an issue to communicate with the newest protocol coming with the SDK version 21.0.0.0. 
If you are using Windows, could you try to use v20.0?
You can download it here: https://github.com/Fluigent/fgt-SDK/releases/tag/20.0.0.0

It should work properly with your MFCS, do not hesitate to inform us if this wasn't the case.

We will keep you in touch as soon as we have solved the problem and released a fix.

Sorry for that,

Cordially,

Mikael

medinahamidovic, 2021-05-27:
Hallo Mikael,

yes, with old SDK 20.0.0 it is working. 
Did you find a solution for 21.0.0?

Medina

mtrellet, 2021-05-27:
Hello,

Nice to hear that you could make it work with the former version. 
We are currently in the last test phase for the new version that will fix the issue with the MFCS. We should release it in the coming days.

Mikael

mtrellet, 2021-06-11:
Hello,

The latest release should fix the issue, please tell us whether this is the case and do not hesitate to re-open the issue if you still encounter the same problem.

Cordially,

Mikael

closed 2021-06-11 by mtrellet