Fluigent/fgt-SDK #32: aarch64 support

mfreer, 2021-05-20:
Hello,

I noticed in the low-level wrapper for the SDK there are checks to determine if the system has the 'aarch64' architecture. However, there are currently no compiled libraries for these types of systems. Are there plans to include these libraries in a future release, and if so, is there an estimated timeframe for release?

Thanks!

mtrellet, 2021-05-21:
Hello,

First, thanks for your interest in our SDK! 
The support of ARM-64bits architecture is not planned yet. However, we will investigate this possibility and keep you in touch if such development is possible and within which timeframe we would be able to release a new version with including this support.

Have a nice day.

mtrellet, 2021-06-23:
Hello,

Thanks for taking time to show your interest. As stated before we do consider this support but our time is split between several projects and we cannot address this issue right now. 
But be sure that this is in our pipeline and part of the very next development efforts. We hope to give you a precise timeframe within the next 2 months at most.

eemerson89, 2021-08-23:
Hi there, I was wondering if you all had been able to figure at time frame for working on this development effort? Thanks!

mtrellet, 2021-11-19:
Hello there,

As you might have seen we have just released a [version](https://github.com/Fluigent/fgt-SDK/releases/tag/21.3.0.0) of the SDK that supports ARM architectures. We could only test it on a Raspberry Pi 4 so far, any additional feedback will be greatly appreciated!

Thanks in advance.

eemerson89, 2021-11-23:
Thank you! We'll take a look at this update with our Jetson Nano and report back what we find.