Fluigent/fgt-SDK #34: fgt-SDK glibc version dependency limits compatibility

william-richards-idexx, 2021-06-22:
Hi all! 

Would it be possible to build these against older versions of glibc and gcc for wider compatibility of distribution?

I'm trying to load up the fgt-SDK on a NI CompactRIO, which runs a Debian fork, and it's failing to load up some dependencies.
```
admin@NI-cRIO-9033:/usr/local/lib# ldd libfgt_SDK.so
./libfgt_SDK.so: /lib/libm.so.6: version `GLIBC_2.29' not found (required by ./libfgt_SDK.so)
./libfgt_SDK.so: /lib/libpthread.so.0: version `GLIBC_2.30' not found (required by ./libfgt_SDK.so)
./libfgt_SDK.so: /usr/lib/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by ./libfgt_SDK.so)
```

glibc-2.24 is the latest NI is distributing, and the [Debian mainline](https://tracker.debian.org/pkg/glibc) lists glibc-2.28-10 as the latest stable. gcc-7.3.0-r.071 is the latest stable version I have in my package feeds. (That would mean GLIBCXX_3.4.24.) 

I know Ubuntu is a few versions ahead, and meets those versions, unfortunately some other distros have slower approval cycles.

Yes, I am asking NI if they have those updated versions in pipeline. For a lab/experimental environment manually installing unstable/untested glibc might not be a problem, but this is for production so I'm a bit hesitant to go that route.

mtrellet, 2021-06-23:
Hello,

Thanks for taking time to raise your issue and for your interest in our SDK. After some investigations and changes we have successfully built a version of the lib with lower requirements. 
We did not have time to test it thoroughly though, would you mind giving it a try and report any problem?

You can find the new version in this temporary branch: https://github.com/Fluigent/fgt-SDK/tree/compact-rio-compatibility-test

Thanks in advance!

william-richards-idexx, 2021-06-23:
Thank you for the quick response/release!

So far I've loaded it up on the cRIO-9033, and can successfully init(), close(), and detect(), it returns the correct serial number from the MFCS_EZ I hooked up to it. So initial try is a success. I will port more of the driver and keep you updated if I see any problems.

I have a few LineUP modules I could try it out with later, but MFCS_EZ is the priority for me.

closed 2021-11-19 by mtrellet