Fluigent/fgt-SDK #43: How does one set flowRate using the fgt-SDK commands?

Immaad-Mir, 2022-06-24:
Hello,

I am trying to replace some old legacy code with the fgt-sdk commands.
I am having trouble figuring out how to set flow rate using these commands.
Is there a command to do this?

Best,

Immaad

Ygor-Oliveira, 2022-06-27:
Hello,

You can call the function [fgt_set_sensorRegulation](https://raw.githubusercontent.com/Fluigent/fgt-SDK/41aeb58fb442e795e2c365d0ad3d7e2b9e1bd5a3/Fluigent%20SDK.pdf#subsection.5.2.17) to set the flow rate. It will continue to be regulated until canceled with a direct command to the pressure controller. To find the indices of the pressure controller and sensor, you can use the functions [fgt_get_pressureChannelsInfo](https://raw.githubusercontent.com/Fluigent/fgt-SDK/41aeb58fb442e795e2c365d0ad3d7e2b9e1bd5a3/Fluigent%20SDK.pdf#subsection.5.2.10) and [fgt_get_sensorChannelsInfo](https://raw.githubusercontent.com/Fluigent/fgt-SDK/41aeb58fb442e795e2c365d0ad3d7e2b9e1bd5a3/Fluigent%20SDK.pdf#subsection.5.2.11), respectively. If there is only one pressure controller and one sensor connected, setting both indices to 0 also works.

Best regards,
Ygor

Immaad-Mir, 2022-06-27:
Hello Ygor,

Thats what I initially thought but it still does not work-- no matter what I set setPoint to the flow rate is always 0ul/min. 
Whats weird is all of the other fgt commands work as expected. 

Immaad

Ygor-Oliveira, 2022-06-28:
Hello,

Does the function return any errors? What types of instruments are you using? Have you confirmed that the flow rate value you requested can be reached with a pressure command? If you set the pressure to some value manually and then set the flow rate, does it change the pressure back to 0?

Note that if the flow rate sensor is connected in such a way that increasing the pressure causes the measured flow rate to decrease, you will have to invert the measurement for the regulation to work. If it is not possible to physically invert the sensor, you can do so by calling [fgt_set_sensorCustomScale](https://raw.githubusercontent.com/Fluigent/fgt-SDK/41aeb58fb442e795e2c365d0ad3d7e2b9e1bd5a3/Fluigent%20SDK.pdf#subsection.5.2.31) and setting (a, b, c) to (-1, 0, 0). You can also use the [fgt_set_customSensorRegulation](https://raw.githubusercontent.com/Fluigent/fgt-SDK/41aeb58fb442e795e2c365d0ad3d7e2b9e1bd5a3/Fluigent%20SDK.pdf#subsection.5.2.34) function instead for the regulation, and give it the sensor measurement and setpoint already inverted.

Ygor

Ygor-Oliveira, 2022-06-28:
Which programming language are you using? Could you provide a minimal sample of your code showing the problem?

Immaad-Mir, 2022-06-29:
Hello,

The programming language that I am using is Python.
Here is a short snippet of code where I would expect the code to switch valve index 0 to valve position 3 then set the flow rate to 300ul/min. fgt_set_valvePosition() works and the valve position is changing but fgt_set_sensorRegulation is not changing the flow rate to 300. 
```
from Flugent.SDK import fgt_init, fgt_close
from Flugent.SDK import fgt_SWITCH_DIRECTION
from Flugent.SDK import fgt_set_valvePosition, fgt_set_sensorRegulation
fgt_init()
fgt_set_valvePosition(0,3,direction=fgt_SWITCH_DIRECTION.Anticlockwise, wait=5)
time.sleep(2.5)
fgt_set_sensorRegulation(0,0,300)
fgt_close()
```

Does  it look like I am missing anything?

Ygor-Oliveira, 2022-06-30:
`fgt_close()` immediately closes the connection to all the instruments and stops all running threads, including the flow rate regulation. So this code is stopping the regulation before it has a chance to run. You should call `fgt_close()` only when the experiment is finished, to release the resources and avoid errors when exiting the program.

 Note that the flow rate regulation always runs in the SDK process, not in the instruments, since it can associate instruments that cannot communicate directly with each other. This means that you will not see the flow rate setpoint appear on the Flow EZ screen after calling `fgt_set_sensorRegulation`. It will instead continuously send pressure commands to the chosen pressure controller in order to maintain the desired flow rate on the chosen flow rate sensor, regardless of whether the two instruments are connected to each other physically.

On another note, when calling `fgt_set_valvePosition` the `wait` parameter should be set to True or False (the default is True). But it will use the usual Python rules to covert any parameter to a boolean, so 5 will be interpreted as True.

mtrellet, 2022-07-19:
@Immaad-Mir Did you have the chance to solve your issue following Ygor's last advices?

Immaad-Mir, 2022-07-21:
Hi,

I was able to implement it but it does not seem to do as we want. This command seems to significantly increase pressure to reach the wanted flow rate, something our old command didn't do. With the pressure too high it can ruin our system.

mtrellet, 2022-07-22:
Hello,

Would you mind providing a snippet of what your old code was doing? We might be able to spot where the difference(s) lie.
Few advices in the meantime:

- Play with the response time of your instruments using those two functions:
[image]
- Set an upper limit for your pressure source to prevent any overpressure for your system
[image]

Hope this will help.

Immaad-Mir, 2022-07-25:
Hi,

My stripped down version of my current code, that does not include use fgt_set_sensorRegulation to set flowrate looks like this:
```
from LineUP_Middleware import *
from LineUP_lowLevel import *
from Fluigent.SDK import fgt_init, fgt_close, fgt_set_valvePosition

LineUP = LineUPClassicalSessionFactory().Create(0)

def set_flowrate(buffer: str, flowrate: int, wait_time: int):
    fgt_init()
    fgt_set_valvePosition(0,3,direction=fgt_SWITCH_DIRECTION.Anticlockwise, wait=5)
    LineUP.SetFlowrate(0, flowrate)
    # wait time also happens here
    if exit_event.wait(timeout=wait_time): #can force exit application
        print("Force exiting.")
        return
    print('Completed wait time for set flow rate.' + str(datetime.datetime.now()))
    fgt_close()

     
```

When called this method essentially sets the flowrate to what ever integer value is stored in the "flowrate" variable.
As an example using this current code without the fgt-sdk commands if we set the flowrate to 500 we see the pressure fluctuate between 60 and 70mbar which is ideal.

mtrellet, 2022-08-03:
Hello,

I don't see anything that you would not be able to replicate with the newest versions of the SDK. Did you try playing with the functions I've attached in my previous comment? Did it solve your problem?

mtrellet, 2022-11-15:
Hello @Immaad-Mir 

Can we consider this issue as closed?

Best

closed 2022-12-20 by mtrellet