Fluigent/fgt-SDK #49: Question about the PX1 pressure controller

eeycn, 2022-10-06:
I have a question about the pressure controller. Does the pressure controller support CAN communication protocol? Besides the hardware, is there any difference in the software, e.g. the command? Could I use the same Python code with CAN communication?

eeycn, 2022-10-06:
> Good morning,
> 
> Unfortunately no, the CAN communication protocol is not supported by the pressure controller (and any of our modules for that matter). No CAN bus is available and can be connected to.
> 
> Best!

Could I use an RS232 to CAN converter to control the PX1?

mtrellet, 2022-10-08:
Hello,

This might work but it will greatly depends on the RS232/CAN adapter. CAN allows for a maximum of 64 bits data length in the messages while the RS232 commands are much longer. 

Hope this helps.

closed 2022-11-15 by mtrellet