Fluigent/fgt-SDK #50: Aria C# Integration

critter4x, 2022-10-07:
My lab has to control multiple different instruments with C# and was wondering if Aria was compatible with the software development kit.

mtrellet, 2022-10-08:
Hello,

Unfortunately not as the moment, Aria can only be controlled via its dedicated software. 
Communication with other modules can be done via TTL signals only but we plan to expend the communication capabilities with other protocol(s) in the near future.

Best

critter4x, 2022-10-10:
If integration with C# isn't possible, would you by chance know a way to send TTL signals via desktop to the Aria? If I could set up a system where the desktop and Aria can send ready signals back and forth that would work great too.

FINCMAR, 2022-10-10:
Hello,

 

The Aria has two built in TTL ports which can be configured in the Aria
software to send or receive TTL signals. This is described in the User's
Manual. If you need assistance, in the US please reach out to my colleague
Jason ***@***.***
***@***.***> ) for assistance. If you are outside
the US, please let me know the country and I will refer you to the local
person.

 

Regards,

 

Robert Pelletier

Fluigent, Inc.

73 Princeton St., Suite 310

N. Chelmsford, MA 01863

Direct: 978 306-6985

Mobile: 978 399-4390

 <http://www.fluigent.com/> www.fluigent.com 

 

<https://github.com/Fluigent/fgt-SDK/issues/50#issuecomment-1273795012> view
it on GitHub, or
R4WHANCNFSM6AAAAAAQ73C3GM> unsubscribe.

This email has been checked for viruses by AVG antivirus software.
www.avg.com

closed 2022-11-15 by mtrellet