Fluigent/fgt-SDK #51: Missing .vip file for LabVIEW Installation

git-TERA, 2022-10-25:
Hello!

Im trying to use the Fluigent SDK to control our Fluigent equipment using LabVIEW 2020.  According to the 'Fluigent SDK.pfd', I should be using a .vip file to install the SDK using the National Instruments Package Manager.  However, it appears as though the necessary .vip file is missing from the repository.  I was wondering if it could be added to the repository?

Thank you!

mtrellet, 2022-10-25:
Hello!

The latest release does contain the required `.vip`. Are you sure you've downloaded the [release](https://github.com/Fluigent/fgt-SDK/releases/tag/22.1.0.0) and not the source code instead..?

Best regards

git-TERA, 2022-10-25:
Thank you!  I downloaded the source code, not the release.  Thank you for your help!

closed 2022-10-25 by git-TERA