Fluigent/fgt-SDK #52: Can devices be awoken from sleep?

jsakkos-curio, 2022-10-27:
Our M-switch gave error code 5: "Module is in sleep mode, orders are not taken into account". Is there a command to wake it up?

Ygor-Oliveira, 2022-10-28:
Hello,

You can awaken the instrument using the [fgt_set_power](https://raw.githubusercontent.com/Fluigent/fgt-SDK/22.1.0.0/Fluigent%20SDK.pdf#subsection.5.2.43) function, setting the `powerState` input to 1. It must be called on the controller (e.g., the LineUP Link), not the individual valves.

Best,

jsakkos-curio, 2022-11-02:
Thanks for the tip, @Ygor-Oliveira. I've written a script to extensively test our switches. What we observed is that sometimes the switches still respond with error code 0, but don't physically actuate. We've been able to catch these errors by timing the actuation duration. Additionally, waking from sleep does not always work. Instead, powering off the Push-Pull and powering it back on seems to resolve some additional cases.

````from Fluigent.SDK import fgt_init, fgt_close
from Fluigent.SDK import fgt_SWITCH_DIRECTION
from Fluigent.SDK import (
    fgt_set_valvePosition,    
    fgt_set_power,
    fgt_set_sensorRegulation,
)
from tqdm import tqdm
import sys
import time
def run():  
    fgt_init()
    
    for i in tqdm(range(1000)):
        for j in range(10):
            t1=time.time()
            error_code=fgt_set_valvePosition(0,j, direction=fgt_SWITCH_DIRECTION.Anticlockwise, wait=1)
            t2=time.time()
            time.sleep(2.5)
            verify_switch_actuation(0,j,error_code,t1,t2)
            

    fgt_close()
def verify_switch_actuation(switch: int, position: int, error_code, t1, t2):
    """Check that M-switches actuate as expected via error code and the switching duration. For durations less than 0.5 seconds but not 0, the switch has likely failed to actuate and requires waking from sleep.

    Args:
        switch (int): Switch selection, 0 or 1.
        position (int): Switch position, 0 to 9.
        error_code (_type_): Error code returned by the switch after actuation. See Fluigent SDK for list of error codes.
        t1 (float): Start time
        t2 (float): End time
    """
    # Check error code and actuation time
    if error_code.__dict__["_fgt_ENUM__value"] != 0 or (
        (t2 - t1) > 0 and (t2 - t1) < 0.5
    ):
        print(f"M-switch error on {switch} {position}, attempting to wake up.")
        error_code = fgt_set_power(0, 1)
        t1 = time.time()
        error_code = fgt_set_valvePosition(
            switch, position, direction=fgt_SWITCH_DIRECTION.Anticlockwise, wait=5
        )
        t2 = time.time()
        time.sleep(2.5)
        if error_code.__dict__["_fgt_ENUM__value"] != 0 or (
            (t2 - t1) > 0 and (t2 - t1) < 0.5
        ):
            print("Unable to wake M-switch from sleep. Rebooting controller.")
            # Turn off LineUP Controller
            error_code = fgt_set_power(0, 0)
            time.sleep(5)
            # Turn on LineUP Controller
            error_code = fgt_set_power(0, 1)
            time.sleep(5)
            fgt_close()
            error_code = fgt_set_valvePosition(
                2, 1           )
            t1 = time.time()
            error_code = fgt_set_valvePosition(
                switch, position, direction=fgt_SWITCH_DIRECTION.Anticlockwise, wait=5
            )
            t2 = time.time()
            time.sleep(2.5)
            # Reset 2-switch to open

            if error_code.__dict__["_fgt_ENUM__value"] != 0 or (
                (t2 - t1) > 0 and (t2 - t1) < 0.5
            ):
                print("Unable to wake M-switch. Stopping flow.")
                fgt_set_sensorRegulation(0, 0, 0)
                # exit_event.set()
                sys.exit("M-switch not actuating, exiting...")
    else:
        pass
        #print('Actuated correctly')
if __name__ == "__main__":
    run()

closed 2022-11-02 by jsakkos-curio