Fluigent/fgt-SDK #53: Difficulty Regulating Multiple Flowrates

cytonomembarno, 2022-11-02:
All,
I'm using a MFCS pressure board with two FRP sensors (Flow_L_Dual and Flow_M_Dual). I am trying to set two flowrates by calling fgt_set_sensorRegulation, binding each of the sensors to the respective channel. What seems to be happening is the closed-loop regulations compete with each other. I've referred to the documentation and it says that calling fgt_set_sensorRegulation on multiple lines is fine. However, what I'm seeing is an oscillation in each flowrate that seems to correspond to the other. When flowrate-1 pops over the setpoint, flowrate-2 is dipping down.

When I use the Fluigent A-i-O (All-in-One) tool to manually setup the same flowrates, everything behaves as expected- without oscillations. This leads me to believe that either A-i-O is doing something on top of just calling fgt_set_sensorRegulation, or there's something else that I should be doing in my code with either initialization, or steps for starting up the regulation (although the Basic Sensor Regulation example seems to do the same steps as me).

snippet of my init:
        ...
	uint16_t sn[FGTSIZ];
	fgt_INSTRUMENT_TYPE it[FGTSIZ];
	// get hw serials
	if (Fgt_detect(sn, it))
	{
		Fgt_init();
		// Pressure
		uint8_t pc(0);
		Fgt_get_pressureChannelCount(&pc);
		fgt_CHANNEL_INFO pi[FGTSIZ];
		Fgt_get_pressureChannelsInfo(pi);
		for (int k = 0; k < pc; k++)
		{
			Fgt_set_pressureUnit(k, "mbar");
			Fgt_set_pressureResponse(pi[k].indexID, 1);
			Fgt_calibratePressure(pi[k].indexID);
		}
		// Sensors
		uint8_t sc(0);
		Fgt_get_sensorChannelCount(&sc);
		fgt_CHANNEL_INFO si[FGTSIZ];
		fgt_SENSOR_TYPE st[FGTSIZ];
		Fgt_get_sensorChannelsInfo(si, st);
		for (int k = 0; k < sc; k++)
		{
			Fgt_set_sensorRegulationResponse(si[k].indexID, 6);
			Fgt_set_sensorUnit(si[k].indexID, std::string("mL/hour"));
			Fgt_set_sensorCalibration(si[k].indexID, fgt_SENSOR_CALIBRATION::H2O);
		}
        ...
snippet of my set's:
        ....
        fgt_set_sensorRegulation(0, 1, 14.0);
        fgt_set_sensorRegulation(1, 0, 1.0);
        ...
If anyone has any advice please let me know.

Thanks,
Mike

mtrellet, 2022-11-04:
Hello Mike,

Sorry for the late answer, your issue raised several discussions here and I wanted to have as much insights as possible before replying.
First, if you don't mind, would it be possible to get a bit more information on the setup and your observations?
- Do we agree that the 2 channels you are regulating are independent from each other? They are not linked at any moment?
- What is the amplitude of the flowrate oscillations you observe? Would you have a snapshot or a record file of the behavior?

Either send details here or, if you prefer, you can send them by email via our support team: support@fluigent.com

Thanks in advance.

cytonomembarno, 2022-11-07:
[image]
Here is a screenshot of the 2 flowrates set to 14.0 mL/h and 1.0 mL/h respectively. The visual is of A-i-O, but the flowrates were set through an app I made. The flowrate oscillations do seem to be correlated from the zoomed in view on a mini dialog I made to view it. As for the setup, The 2 flow channels both join at a microfluidic chip, so the aren't totally independent of one another. I understand that this joining makes pushback seem obvious, but when we run directly out of the A-i-O the oscillation seen it this picture doesn't occur (or is at least much smaller).

mtrellet, 2022-11-08:
Thanks for those information, they will be very helpful to try understand your issue.
May I ask something while we review those new data?
Can you give a try with OxyGEN, our UI software? It relies on the SDK to control our instruments, so you should observe the same behavior. The advantage is that you'll have some records created, both from the instruments output values and their behaviors. Consequently we could maybe see whether something odd is happening.
You can download the software, for free, from this page: https://www.fluigent.com/resources-support/support-tools/software/oxygen/

Once installed and run, you will find records in your `C:\Users\XXX\AppData\Fluigent\OxyGEN` folder. Most interesting data lying in `AppLogs` and `Records` folders. You choose to either create a protocol or to simply live monitor the instruments. Idea would be, for the latter, to set a regulation of your MFCS channels by your FU via the interface. 

To do so, first create your setup:
[image]
And then choose the regulated pressure channel from the list:
[image]
Finally simply click the "Play" button to start regulating the pressure via the selected FU.

Don't forget to click on the "Camera" button to start recording the session:
[image]

Once done, you'll find the records in the `Records` folder mentionned earlier in this post.

Thanks in advance

cytonomembarno, 2022-11-09:
The OxyGEN GUI proved to be a valuable tool while further debugging this issue. I don't think the two loops are actually competing with one another. What seems to happen is when my app starts up, whatever pressure is running gets set to the 0 point. In other words, if the sample pressure is 600 mbar and my app starts, that becomes the new origin, so when I then set the pressure to 600 mbar it was actually be at 1200 mbar. Of course this wasn't immediately apparent when looking in A-i-O because the offset was applied there as well. The OxyGEN App has the Reset Pressure button on the menu bar as well as individually for each device. That wound up being the fix to the issue, however I'm still somewhat unclear as to what my app is doing on startup to cause this behavior. I will continue to poke around in my process, but any insight you may have on what can cause this would be greatly appreciated.

These are the functions used in my initialize function:
Fgt_detect, Fgt_init, Fgt_get_pressureChannelCount, Fgt_get_pressureChannelsInfo, Fgt_set_pressureUnit, Fgt_set_pressureResponse, Fgt_calibratePressure, Fgt_get_sensorChannelCount, Fgt_get_sensorChannelsInfo, Fgt_set_sensorRegulationResponse, Fgt_set_sensorUnit, and Fgt_set_sensorCalibration

Note: the *calibrate functions were my first guess, but commenting them out didn't seem to do anything.

Thanks for all the help,
Mike

Ygor-Oliveira, 2022-11-09:
Hello,

The [Fgt_calibratePressure](https://raw.githubusercontent.com/Fluigent/fgt-SDK/22.1.0.0/Fluigent%20SDK.pdf#subsection.5.2.33) function is the only one that can change the zero of the pressure controller. Here is what it does in the case of the MFCS:

- Close the channel's inlet valve to let the output depressurize.
- Wait until the measured value is constant, with a timeout at 7 seconds.
- Set the pressure sensor's zero to the latest measured value. The value is saved in the instrument's memory and persists after it is powered off.

Normally this should be enough to set the zero to the atmospheric pressure, but if you have a large reservoir that takes long to depressurize, it might be better to only calibrate the MFCS when its output is open to the atmosphere. In any case, the sensor does not need to be calibrated often, as it only drifts slowly over the years and the flow rate regulation will make up for it.

It is odd that the pressure calibration would affect the flow rate regulation, but it is possible that it gets confused if the measurements are very far from reality. Let us know if the oscillation problems got resolved after fixing the calibration. If not, you can also try increasing the response time you set with the [Fgt_set_sensorRegulationResponse](https://raw.githubusercontent.com/Fluigent/fgt-SDK/22.1.0.0/Fluigent%20SDK.pdf#subsection.5.2.39) function. It is meant to help improve stability in cases where there is dead time between the pressure and the flow rate (due to the elasticity of the chip), or when multiple channels are connected to the same fluidic path and competing with each other. Its implementation is slightly different between A-i-O and the SDK, so the same value might not work for both.

Best,
Ygor

mtrellet, 2022-11-14:
Hello @cytonomembarno ,

Did you have the opportunity to test the last advices from my colleague? Did it improve things or not at all? 
Do not hesitate to report any (new) problems!

Best,
Mikael

cytonomembarno, 2022-12-21:
@mtrellet 
@Ygor-Oliveira 

So the info provided about Fgt_calibratePressure did indeed confirm some of what I was think was happening. The issue still persists. I am wondering if:
"Set the pressure sensor's zero to the latest measured value. The value is saved in the instrument's memory and persists after it is powered off."

is playing a role. I set pressures to 0 to break regulation loops on exit before calling Fgt_close(), but on the next run the oscillations come back. I say *come back* because I have a trick to make them stop - When my app is running and the regulation loops have a sinusoidal instability I can open OxyGEN and they go away. So when OxyGEN starts up it is doing something that seemingly frees up a collision, but I don't know exactly what that is. I've been in a button in my own app to re-calibrate, but it doesn't seem to fix it. In my initial post above I show the steps of my init function where the calibrations happen on start up. Is there any chance something is out of order or missing because it seems as though the OxyGEN GUI is doing something different.

Ygor-Oliveira, 2022-12-21:
Hello,

When OxyGEN starts, it reapplies the same instrument parameters that were active in the previous session. That includes the regulation response time. You could try to open the instrument settings using the gear icon in the live control pane and see what response time it is applying, and set the same value in your script using the function I mentioned in the previous comment. Let us know if that works.

Best,
Ygor

cytonomembarno, 2022-12-21:
I will run some tests now. Thanks for the quick response!

Ygor-Oliveira, 2022-12-22:
Hello again,

I am sorry for my inaccurate reply from earlier. I hadn't realized that the sensor regulation response is stored in the process that is running the SDK and not sent to the instrument, so changing it from OxyGEN would not affect you app. I ran a debugger to get a list of all SDK functions being called when OxyGEN starts, and I think these are the only ones that can have an impact outside the OxyGEN process itself:

```
fgt_init()
fgt_set_sensorCalibration(807359488, 1)
fgt_set_pressureResponse(273375744, 5)
```

`fgt_init()` queries the instrument for its parameters, which causes it to skip a few measurements while replying, but it shouldn't have a big impact, `fgt_set_sensorCalibration()` would change your sensor's calibration table if you were not using the default one, but it seems that you are. However, `fgt_set_pressureResponse()` might be what you are looking for. It actually gets sent to the instrument, unlike the sensor response, and it affects the responsiveness and stability of its pressure control. You should generally set this value according to the size of your reservoir, but the default value of 5 works for most typical cases, so you can try that. I am sorry I didn't notice this earlier.

This does not preclude trying to increase the sensor response time as well, as the Y setup is a difficult case for this algorithm, and you might have to make it go slower to improve stability. If the settling time becomes too slow, you can try to start with a low value and increase it after a few seconds or once the target flow rate is reached.

Best,
Ygor

cytonomembarno, 2022-12-22:
Thank you for that! It's funny you mention these things because I started screwing around with increasing the fgt_set_pressureResponse and it looks as though that is smoothing things out. We've been having some other unrelated issues on the system, so I haven't been able to fully confirm anything. Nonetheless, this info is very helpful, so thank you again.

mtrellet, 2023-04-05:
Hello @cytonomembarno May I close this issue?

closed 2023-04-20 by mtrellet