Fluigent/fgt-SDK #54: Periodic output

ThomasQuenan, 2022-11-15:
Hello,

I am trying to set a periodic pressure output with the Fluigent MFCS-EZ, using the Python library. I can't seem to reach clean sinusoidal waves with frequency around 1 Hz by only implementing the set_pressure fonction in my loop. What would you recommand ? Is there a smart way to do it and what would be the hard/software limitations of such an application ?

Best,
Thomas

mtrellet, 2022-11-15:
Hello @tyquenan 

Setting a periodic pressure is possible but, as you mentionned, will have some limitations depending on the amplitude of the wave you want to reach and its period. 

Important point:
> ... I can't seem to reach clean sinusoidal waves ...

Do we agree that you are looking at the pressure values? Or do you look at the flowrate induced by your pressure changes?

Can you provide some more details about your implementation? What are the intermediate pressure setpoints you are providing? I understood that you update the pressure setpoint every second, it seems to me rather slow for pressure regulation. I would increase the frequency since the pressure tends to quickly reach the setpoint you are providing. 

However, if you want a flowrate control at the end (might not be the case but just in case), I would highly recommend to use the dedicated function to do so (`fgt_set_sensorRegulation`). You can find a nice example here: [Python/examples/BasicSensorRegulation.py](https://github.com/Fluigent/fgt-SDK/blob/master/Python/examples/BasicSensorRegulation.py)

Hope this will help,

Mikael

mtrellet, 2022-12-14:
Hello @tyquenan ,

Can we assume your issue is fixed? If yes I will close this issue.

Thank you.

closed 2022-12-20 by mtrellet