Fluigent/fgt-SDK #55: Communication Protocol

eeycn0128, 2022-11-15:
I have a pressure controller PX1. Where could I find the communication protocol, e.g. command for writing and reading info?

mtrellet, 2022-11-15:
Hello,

Please find below the manual datasheet where are listed the commands used to control a PX via the RS232 protocol. Note that it is also available on our website, in the product page: https://www.fluigent.com/industrial/industrial-products/standard-industrial-components/fluigent-px/)

Manual datasheet: [px_manual_datasheet_v3.pdf](https://github.com/Fluigent/fgt-SDK/files/10016235/px_manual_datasheet_v3.pdf)

If you want to control the PX via USB, then follow the SDK manual and use it as you would do for an MFCS.

Hope this helps

mtrellet, 2022-12-14:
Hello @eeycn0128 ,

Can we assume that my message answered your questions? If yes I will close this issue.

Thank you.

eeycn0128, 2022-12-14:
Hi, 

Is there any command for initialization?  I didn't find it in the manual.

eeycn0128, 2022-12-14:
@mtrellet 
Hi,

Is there any command for initialization? I didn't find it in the manual.

closed 2023-04-05 by mtrellet

mtrevisiol, 2024-03-25:
I am controlling the PX-345 through the RS-232 port.  I can set and query successfully.  I do not see a command to Read the current pressure setpoint.  Alpha can be set and checked, pressure SP cannot (that I can see).

The query equivalent of:
CHAN:1P:XXX

I tried the question mark, no success.

Mark

mtrellet, 2024-03-26:
This is a closed issue that is different from your own issue. I've opened a new issue where we can continue the discussion: https://github.com/Fluigent/fgt-SDK/issues/83