Fluigent/fgt-SDK #57: Sensor calibration not applied

youenv, 2022-12-14:
Hello,

We have a python code that manage several F-OEM (pressure controllers, valves, sensors).

Part of the code is used to change the calibration of the sensor. We can change from water to IPA. But we can used the HFE, OIL or FC40 calibration. When we use this calibration, nothing happens.

Do you have any idea on how to fix this ?
*Fluigent.fgt_set_sensorCalibration(self.Stm.FlowRateM.id, Fluigent.fgt_SENSOR_CALIBRATION.IPA)  => works
*Fluigent.fgt_set_sensorCalibration(self.Stm.FlowRateM.id, Fluigent.fgt_SENSOR_CALIBRATION.HFE)  => does not work

Kind regards,
Youen

Ygor-Oliveira, 2022-12-16:
Hello,

I would like to know the version of the INT-FOEM board (the Controller) as well. I am testing with an INT-FOEM v1.02 and a pressure module v1.01 with the Python SDK version 22.1.0, and I cannot reproduce the issue. It works correctly with all 5 calibration tables. However, there are some differences in how the calibration table is retrieved based on the controller version, so there might be some confusion there, especially if the SDK is ignoring the command without any error messages.

Best,
Ygor

youenv, 2022-12-16:
Here are the information relative to the board :
`Controller info at index: 0
<class 'Fluigent.SDK.fgt_CONTROLLER_INFO'>
SN: 12140
Firmware: 1.02
index: 1928773632
InstrType: F_OEM`

Best,

youen

mtrellet, 2022-12-16:
Hello Youen,

Our current investigation does not allow us to replicate your issue. Just to be sure, are you using the very last version of the SDK (`v22.1`)? And what OS are you using?
To get the version:
```python
from Fluigent import SDK

print(SDK.__version__)
```
We will certainly raise this issue to the support team to try find a solution as soon as possible.

Thanks in advance and sorry for the inconvenience,

Mikael

Ygor-Oliveira, 2022-12-19:
Hello,

That is probably the problem. We had to make some changes to how the calibration table is encoded by the firmware in order to support our new sensor ranges M+ and L+, and those changes affected the original range M as well. The SDK version 22.1.0 is aware of those changes, but version 22.0.0 is not. Could you try the [latest release](https://github.com/Fluigent/fgt-SDK/releases/tag/22.1.0.0) and see if it solves the issue? Thank you.

Best,
Ygor

Ygor-Oliveira, 2022-12-19:
Hello,

Good to know that the calibration problem is solved. We will try to improve our communication about which versions of the SDK are compatible with which firmware versions going forward.

Unfortunately, it is not possible to set or reset the flow rate sensor's zero. The underlying hardware does not offer this functionality. Sorry for the inconvenience.

Best,
Ygor

closed 2023-04-05 by mtrellet