Fluigent/fgt-SDK #60: Intermittent issues with setting flowrate

jsakkos-curio, 2023-01-06:
Using the FLU-L, we control flow with the `fgt_set_sensorRegulation` function. Occasionally, when we set the flowrate to something like 500 uL/min with `fgt_set_sensorRegulation(0,0,500)` , the unit will reach an arbitrary intermediate value, e.g., 180 uL/min, and then stabilize there. Issuing the command a second time resolves this error, but is not always practical. Any idea what could be going on?

mtrellet, 2023-01-09:
Hello @jsakkos-curio,

To help us find the issue and replicate it, do you observe a particular pattern when this error occurs? Is there a set of commands you run before that might trigger the error systematically for instance? Or does it seem completely random? What is the starting flowrate value before you send the `fgt_set_sensorRegulation`?

Another question: Does it happen only for 500uL/min, other values as well? I understood that the intermediate values are random though, right?

Of course, if you have a snippet of your code where the issue happens that would be even nicer.

Thanks in advance and sorry for the invonvenience,

Mikael

jsakkos-curio, 2023-01-09:
@mtrellet,

The only thing I can think of is that it was the first set flow command issued to the unit after initializing. The starting flowrate was not set yet, but it was around 0.

I cannot confirm whether it only happens at 500 uL/min. The intermediate value did seem random.

The code we were using was something like this:

```
    
for position in [0,1,2]:
  current_position = fgt_get_valvePosition(0)
  if current_position != position :
      error_code = fgt_set_valvePosition(
          0, position, direction=fgt_SWITCH_DIRECTION.Shortest, wait=True
            )
  fgt_set_sensorRegulation(0, 0, 500)
  time.sleep(10)
  fgt_set_sensorRegulation(0, 0, 0)
  time.sleep(10)

```

Jonathan

mtrellet, 2023-01-10:
Ok, that's indeed rather strange.

Another question came to my mind, what is the behavior of the pressure source? Does it reach any maximum value for instance? Does it only fluctuate to maintain this random intermediate values (e.g. 180ul/min)?

To fully mimic your issue, could you provide some information about your setup (pressure source(s), controller(s), etc.)?

Thanks in advance,

Mikael

Ygor-Oliveira, 2023-01-10:
Hello,

If calling `fgt_set_sensorRegulation` again solves the problem, then the regulation is most likely being stopped for some reason, because if this function is called while the regulation is already running, all it does is update the setpoint (which does not change on its own). But as far as I can tell, the only way for the regulation to stop on its own (without an explicit function call such as `fgt_set_pressure`) is if it detects that the pressure controller has changed to local mode (i.e., by pressing the "Set local" button on the Flow EZ). It is possible that the instrument is incorrectly reporting that it is in local mode, or that the SDK is misinterpreting its status, but we have never observed that in our tests, and it is hard to confirm without seeing the low level data with a debugger.

The other possibility is that the SDK is converting the setpoint incorrectly on the first call, as you assume. We have not observed this bug either. Unfortunately, the current version of the SDK does not have any tools to allow us to see what it is doing internally. To see whether the regulation is still running or not, you could try to physically restrict the flow once it settles, to see if the pressure increases to compensate. That could help us narrow down the source of the problem.

If you could send us the firmware versions of the instruments you are using (or the output of the [BasicGetInstrumentsInfo](https://github.com/Fluigent/fgt-SDK/blob/22.1.0.0/Python/examples/BasicGetInstrumentsInfo.py) example), that would be helpful as well.

Best,
Ygor

jsakkos-curio, 2023-01-10:
@mtrellet, the pressure gets set below what is required for the desired flowrate. It doesn't fluctuate after reaching its target.
@mtrellet, @Ygor-Oliveira, I've attached the output of the BasicGetInstrumentsInfo script:
[fgt output.txt](https://github.com/Fluigent/fgt-SDK/files/10386459/fgt.output.txt)

I've been able to reproduce this issue repeatedly by power cycling the controller:

```
fgt_set_power(0,0)
time.sleep(5)
fgt_set_power(0,1)
time.sleep(5)
fgt_set_sensorRegulation(0,0,500)
```
Jonathan

Ygor-Oliveira, 2023-01-11:
Hello,

We were able to reproduce the issue. There is indeed a logic error in the SDK. The problem is that when the regulation is started with the Flow EZ in local mode, it stops immediately after the first iteration, before the module has a chance to report that it has switched to remote. It will happen every time after a power cycle, because the Flow EZ boots in local mode and only switches to remote when a pressure command is set.

We are going to work on a fix for that. In the mean time, you can add this before the regulation command to ensure the module is in remote mode before the regulation is started:

```py
fgt_set_pressure(0, 0)
time.sleep(0.2)
```

Or, if you would like to keep the module at its current pressure as well:

```py
(pmin, pmax) = fgt_get_pressureRange(0)
current_pressure = fgt_get_pressure(0)
fgt_set_pressure(0, min(max(pmin, current_pressure), pmax))
time.sleep(0.2)
```

Or you can send the regulation command twice with a small delay between them:

```py
fgt_set_sensorRegulation(0, 0, 500)
time.sleep(0.2)
fgt_set_sensorRegulation(0, 0, 500)
```

Thank you for your bug report and sorry for the inconvenience.

Best,
Ygor

jsakkos-curio, 2023-01-11:
Thanks @Ygor-Oliveira. That workaround should be fine for now. We'll test on our end to make sure.

PR #69 (2023-06-16): Update SDK to version 23.0.0.0
### Added
- Support for custom Switchboard with 16 Two-Switch ports
- The SDK can now log events for debugging
- Function `fgt_set_log_verbosity` to enable and disable event logs
- Function `fgt_set_log_output_mode` to set how event logs are returned to the user
- Function `fgt_get_next_log` to retrieve event logs manually as strings

### Changed
- Switchboard ports A to D are now mapped to port numbers 16 to 19, instead of 8 to 11.

### Fixed
- Fixed spurious errors being returned by `fgt_get_sensorStatus`.
- F-OEM no longer returns brief connection errors when  [... cut]
closed 2023-06-16 by Ygor-Oliveira