Fluigent/fgt-SDK #61: PID Tuning?

jsakkos-curio, 2023-01-12:
Is it possible to tune the performance of the PID used in the fgt_set_sensorRegulation or fgt_set_customSensorRegulation functions? Due to larger system volume, it currently takes about 30s to get from 0 to 500 uL/min. We have a Flow EZ and FLU-L.

mtrellet, 2023-01-12:
Hello @jsakkos-curio,

What is the volume of the reservoir(s) you are using? 
You can play with the response time of both your regulation (`fgt_set_sensorRegulationResponse`) and your pressure sources (`fgt_set_pressureResponse`). Both functions are documented with the manual to know what they do and what values are possible.

Hope this will help,

Mikael

jsakkos-curio, 2023-01-12:
@mtrellet,

We use mostly 15 mL reservoirs, but have a few 1 L ones, which substantially affect the performance. I tried using `fgt_set_sensorRegulationResponse` with the minimum response value (2s for Flow EZ), and it only improved the performance about 20%. Generally, our system is also much slower to go from 0 to -500 uL/min, by about 2x.

Jonathan

mtrellet, 2023-01-13:
Ah ok, that's indeed some large reservoirs there. Some new questions then:
- Did you try to manually set the pressure at the maximum on the FEZ and calculate how long it takes to reach the flow-rate you want?
- How long does it take to reach the pressure requested..? Usually, when pressurizing large reservoirs, the FEZ can take some time to reach the pressure order.
- Did you try increasing the input pressure? In your situation, the FEZ is certainly pulsing the air at its max rate, so the input pressure must account for that and provide the pressure difference to cop with it.
- Be careful to optimise the tubing between the FEZ and the reservoir, try to have an inner diameter as large as possible. The length can also play a role (but is less significant than the diameter) so try to keep it as short as possible. This will reduce the resistivity and increase the air flow-rate provided by the FEZ.

Mikael

jsakkos-curio, 2023-01-13:
@mtrellet,

We haven't typically set the pressure, but it was a good suggestion. For comparison, it took 2.1 seconds to go from a flow and pressure of 0 uL/min, -35 mBar to 500 uL/min, 290 mBar by setting the pressure. To go back to 0, it took ~51s. By using the sensor regulation, it took 37s to get to 500 uL/min and 62 to get back to 0 uL/min.

Can we increase the input pressure safely? It's at 1100 mBar, which is what the FEZ says is the maximum input. I'm sure that our pressure source is capable of higher pressure.

Thanks for the tip on the tubing. There are certainly some areas where we could increase the tubing diameter.

Jonathan

mtrellet, 2023-01-19:
Hello @jsakkos-curio ,

Sorry for the late answer, I needed to ask around to get more info on the regulation algorithm in particular cases before replying and the last days were quite packed. 

So I could fetch some more information and there is indeed a limitation of the algorithm when addressing large reservoirs (and consequently high pressure differential commands). When set manually, a large pressure increase is facilitated by ON/OFF valves, in the case of the DFC this is not the case because those valves induce a lot of unstability for the regulation. So only the proportional valves are used but they have a rather low air flow.

A possible workaround in your case would be to give a pressure order close to what you've observed as the average pressure to reach the flow-rate you want. You can set the pressure 10/20% below this average value and then only activate the regulation. (Don't forget to give the FlowEZ some time to reach the required pressure before starting the regulation). This should help a lot and reduce the time it takes to reach your target flow-rate.

Hope this will help, keep us updated whether this works for you,

Best,

Mikael

jsakkos-curio, 2023-01-20:
Thanks for the info @mtrellet.

closed 2023-01-20 by jsakkos-curio