Fluigent/fgt-SDK #63: What does this mean: `LineUP error 2 - Unavailable functionality`

paulmueller, 2023-01-23:
Hi, I am getting these error messages in stdout on Windows.

The setup consists of two flow sensors and two pressure regulators. I am getting this error message, regardless of whether  I set the pressure or internal feedback loop.

```
fgt_get_sensorValue sensor channel 1122731009 of type LineUP error 2 - Unavailable functionality Controller SN: 11960 info: remote control
```

What does the error message mean? Is this related to #60 ?

Ygor-Oliveira, 2023-01-23:
Hello,

This is not related to the other issue. The error you are seeing is due to an internal function call that is incorrectly updating the "last error" field of the channel, so its error message is reported as if it came from the function you called. This is a bug introduced in the latest version of the SDK (22.2.0) that we are going to fix for the next release. The `fgt_get_sensorValue` function itself should return the correct error code and sensor measurement, and you can get rid of the stdout print by calling the `fgt_set_errorReportMode` function and setting it to "None" (it will be either a string or an enum, depending on which language you are using). Let us know if this works for you. Sorry for the inconvenience.

Best,
Ygor

paulmueller, 2023-01-23:
OK thanks, so this means I can safely ignore this error?

Ygor-Oliveira, 2023-01-24:
Yes, the correct error code can be obtained from the function itself, if needed for control flow. For example, in Python:

```py
from Fluigent.SDK import fgt_get_sensorValue, fgt_set_errorReportMode

fgt_set_errorReportMode("None")

err, value = fgt_get_sensorValue(0, get_error=True)
if int(err):
    print(int(err), err)
```

closed 2023-04-20 by mtrellet