Fluigent/fgt-SDK #64: Question on PX1

eeycn0128, 2023-02-03:
I am using the PX1 as a pressure controller in our system. However, I noticed that the actual pressure doesn't go down to 0 after the pressure is set to 0. It usually costs a long time to release the pressure. For example, I set the pressure to 300 mbar for 10s then set the pressure to 0 bar. It may take 15 seconds or more to release the pressure I read the pressure using the fgt_fet_pressure() function. 

Anything wrong with my application?

Ygor-Oliveira, 2023-02-06:
Hello,

What is the volume of the reservoir you are using? The PX depressurizes through a passive leak, so its response time when decreasing the pressure is generally longer than when increasing.

Best,
Ygor

mtrellet, 2023-04-05:
@eeycn0128 Did we answer your question?

If yes, can I close this issue?

closed 2023-04-20 by mtrellet