Fluigent/fgt-SDK #65: Python: fgt_SWITCH_DIRECTION object cannot be interpreted as integer

nfullam-Ginkgo, 2023-04-04:
In the code snippet below, `set_valvePosition` gives a type error. If the default argument for direction is instead passed already cast as an int it works. (`SWITCH_DIRECTION.Shortest` vs `int(SWITCH_DIRECTION.Shortest)`)

```
import Fluigent.SDK as flgt
LineUP_config = [
    # FlowEZ with +-1bar pressure source and NIFS
    1, 100, 0, 5, flgt.fgt_SENSOR_TYPE.Flow_L_NIFS, 0, 0, 0, 0, 0,
    # SwitchEZ with L-switch
    4, 103, 0, flgt.fgt_VALVE_TYPE.LSwitch, 0, 0, 0, 0, 0, 0
]

LineUp_example = flgt.fgt_create_simulated_instr(
    flgt.fgt_INSTRUMENT_TYPE.LineUP,
    1,
    0,
    LineUP_config
)

flgt.fgt_init()
controller_array = flgt.fgt_get_controllersInfo()
pressure_array = flgt.fgt_get_pressureChannelsInfo()
valve_array = flgt.fgt_get_valveChannelsInfo()

this_valve = valve_array[0][0].indexID
ranges = flgt.fgt_get_valveRange(this_valve)
print(flgt.fgt_get_valvePosition(this_valve))
flgt.fgt_set_valvePosition(this_valve, ranges)
print(flgt.fgt_get_valvePosition(this_valve))
flgt.fgt_close()
```
Python 3.10.4
Traceback
```
  File "C:\fgt-SDK\Python\simtesting.py", line 28, in <module>
    flgt.fgt_set_valvePosition(this_valve, ranges, direction= flgt.fgt_SWITCH_DIRECTION.Shortest)
  File "C:\fgt-SDK\Python\Fluigent\SDK\__init__.py", line 719, in fgt_set_valvePosition
    c_error, = low_level_function(valve_index, position, direction, wait)
  File "C:\fgt-SDK\Python\Fluigent\SDK\low_level.py", line 600, in fgt_set_valvePosition
    c_error = c_ubyte(lib.fgt_set_valvePosition(c_uint(valve_index), c_int(position), c_int(direction), c_int(wait_int)))
TypeError: 'fgt_SWITCH_DIRECTION' object cannot be interpreted as an integer
```

Ygor-Oliveira, 2023-04-05:
It seems that implicit conversion via the `__int__()` method was removed in [Python 3.10](https://docs.python.org/3.10/whatsnew/changelog.html#id294). We have to update the enum classes to define `__index__()` instead.

> [bpo-36048](https://bugs.python.org/issue?@action=redirect&bpo=36048): The [__index__()](https://docs.python.org/3.10/reference/datamodel.html#object.__index__) special method will be used instead of [__int__()](https://docs.python.org/3.10/reference/datamodel.html#object.__int__) for implicit conversion of Python numbers to C integers. Using the __int__() method in implicit conversions has been deprecated.

We will fix it for the next release. Thank you for your bug report and sorry for the inconvenience.

nfullam-Ginkgo, 2023-04-06:
Thank you for the swift response.

PR #69 (2023-06-16): Update SDK to version 23.0.0.0
### Added
- Support for custom Switchboard with 16 Two-Switch ports
- The SDK can now log events for debugging
- Function `fgt_set_log_verbosity` to enable and disable event logs
- Function `fgt_set_log_output_mode` to set how event logs are returned to the user
- Function `fgt_get_next_log` to retrieve event logs manually as strings

### Changed
- Switchboard ports A to D are now mapped to port numbers 16 to 19, instead of 8 to 11.

### Fixed
- Fixed spurious errors being returned by `fgt_get_sensorStatus`.
- F-OEM no longer returns brief connection errors when  [... cut]
closed 2023-06-16 by Ygor-Oliveira