Fluigent/fgt-SDK #68: Non updated calibration when changing fluid type in flow sensors ( Python SDK )

JeremieBours0, 2023-05-15:
Hello , and thank you for your work , 

We are working on a machine using some of your sensors , your flow-meters sensors are used to measure the amount of liquid flowing through the experiment.

We are using the Python SDK to get ul/min values in our software , everything works well except when we change the fluid type :

Exemple :
The sensors is first in H2O mode , we can read the correct range from it ( 0-120 )
When we change the fluid type to , let say , IPA , the sensor report the correct fluid name ( IPA ) but shows an non updated range : ( 0-120 )

We except the sensor to give 0-600 for IPA

This issue is not constant : from time to time it works  , mostly not

Any idea what would be the issue ?

Here are some logs to show what is happening :  
```
setting calibration to IPA
OK
[WARNING] fgt_get_sensorCalibration sensor channel 1928774657 of type F_OEM error 14 - Sensor is calibrating Controller SN: 12140 info: remote control
calibration is: NONE
[WARNING] fgt_get_sensorCalibration sensor channel 1928774657 of type F_OEM error 14 - Sensor is calibrating Controller SN: 12140 info: remote control
sensor range :  120
calibration is: NONE
[WARNING] fgt_get_sensorCalibration sensor channel 1928774657 of type F_OEM error 5 - Module is in sleep mode Controller SN: 12140 info: remote control
[WARNING] fgt_get_sensorRange sensor channel 1928774657 of type F_OEM error 5 - Module is in sleep mode Controller SN: 12140 info: remote control
sensor range :  0
calibration is: NONE
sensor range :  120
calibration done:  IPA
sensor range :  120
```
Here is an example on how the code is used : 

````python
    # After user input a fluide type on a sensor onto the GUI
    # send the specific calibration to this sensor
    def actualize_calibrations(self, calibration, number, id):
        print(f'coucou falcon {number}, setting calibration to {calibration}')
        if number == 1:
            self.C_calibration_txt = calibration
            if calibration == 'H2O':
                Fluigent.fgt_set_sensorCalibration(id, Fluigent.fgt_SENSOR_CALIBRATION.H2O)
            elif calibration == 'IPA':
                Fluigent.fgt_set_sensorCalibration(id, Fluigent.fgt_SENSOR_CALIBRATION.IPA)
````

Do not hesitate to ask for some more details , 
Jérémie

Ygor-Oliveira, 2023-05-15:
Hello,

We confirmed that the sensor range sometimes does not update when the calibration is changed. We are investigating the issue and will fix it for the next release. In the meantime, a workaround is to call `fgt_init()` (without arguments) after the instrument returns the updated calibration, to force the SDK to update channel parameters. Sorry for the inconvenience.

Best regards,
Ygor

JeremieBours0, 2023-05-15:
Hello , 

thank you for this fast reply , we will test the workaround, 
if it works in our application , this will be a fine solution ! 

best regards , 
Jérémie

JeremieBours0, 2025-08-01:
Hello , 

It is me again , **we updated to sdk 23.0.0** , 
and we tested the workaround.

here is the code we use in the client machine : 

`    def actualize_calibrations(self, calibration, type):
        print(f"falcon {type}, setting calibration to {calibration}")
        if type == "CTU":
            id_ = self.Stm.FlowRateL_CTU.id
            self.C_calibration_txt = calibration
            if calibration == "H2O":
                Fluigent.fgt_set_sensorCalibration(
                    id_, Fluigent.fgt_SENSOR_CALIBRATION.H2O
                )
            elif calibration == "IPA":
                Fluigent.fgt_set_sensorCalibration(
                    id_, Fluigent.fgt_SENSOR_CALIBRATION.IPA
                )
        elif type == "SHELL":
            id_ = self.Stm.FlowRateM_SHELL.id
            self.S1_calibration_txt = calibration
            if calibration == "H2O":
                print(
                    Fluigent.fgt_set_sensorCalibration(
                        id_, Fluigent.fgt_SENSOR_CALIBRATION.H2O
                    )
                )
            elif calibration == "IPA":
                print(
                    Fluigent.fgt_set_sensorCalibration(
                        id_, Fluigent.fgt_SENSOR_CALIBRATION.IPA
                    )
                )
            elif calibration == "HFE":
                Fluigent.fgt_set_sensorCalibration(
                    id_, Fluigent.fgt_SENSOR_CALIBRATION.HFE
                )
            elif calibration == "FC40":
                Fluigent.fgt_set_sensorCalibration(
                    id_, Fluigent.fgt_SENSOR_CALIBRATION.FC40
                )
            elif calibration == "OIL":
                Fluigent.fgt_set_sensorCalibration(
                    id_, Fluigent.fgt_SENSOR_CALIBRATION.OIL
                )
        elif type == "CORE":
            id_ = self.Stm.FlowRateM_CORE.id
            self.C1_calibration_txt = calibration
            if calibration == "H2O":
                Fluigent.fgt_set_sensorCalibration(
                    id_, Fluigent.fgt_SENSOR_CALIBRATION.H2O
                )
            elif calibration == "IPA":
                Fluigent.fgt_set_sensorCalibration(
                    id_, Fluigent.fgt_SENSOR_CALIBRATION.IPA
                )
            elif calibration == "HFE":
                Fluigent.fgt_set_sensorCalibration(
                    id_, Fluigent.fgt_SENSOR_CALIBRATION.HFE
                )
            elif calibration == "FC40":
                Fluigent.fgt_set_sensorCalibration(
                    id_, Fluigent.fgt_SENSOR_CALIBRATION.FC40
                )
            elif calibration == "OIL":
                Fluigent.fgt_set_sensorCalibration(
                    id_, Fluigent.fgt_SENSOR_CALIBRATION.OIL
                )

        Fluigent.fgt_init() # Fluigent workaround ( github support )
        self.Stm.max_flowrate_CTU = round(
            Fluigent.fgt_get_sensorRange(self.Stm.FlowRateL_CTU.id, False)[1]
        )
        self.Stm.max_flowrate_SHELL = round(
            Fluigent.fgt_get_sensorRange(self.Stm.FlowRateM_SHELL.id, False)[1]
        )
        self.Stm.max_flowrate_CORE = round(
            Fluigent.fgt_get_sensorRange(self.Stm.FlowRateM_CORE.id, False)[1]
        )
        print("max_flowrate_CTU :",self.Stm.max_flowrate_CTU )
        print("max_flowrate_SHELL :",self.Stm.max_flowrate_SHELL )
        print("max_flowrate_CORE :",self.Stm.max_flowrate_CORE )`

I think the error is still present at our side : 

The flowrate is updated , but just after that is reset 
The error 14 still there 

`falcon CTU, setting calibration to H2O
max_flowrate_CTU : 600
max_flowrate_SHELL : 120
max_flowrate_CORE : 1100
[WARNING] fgt_get_sensorValue sensor channel 1934189570 of type F_OEM error 14 - Sensor is calibrating Controller SN: 13462 info: remote control
[WARNING] fgt_get_sensorValue sensor channel 1934189570 of type F_OEM error 14 - Sensor is calibrating Controller SN: 13462 info: remote control
falcon CTU, setting calibration to IPA
max_flowrate_CTU : 120
max_flowrate_SHELL : 120
max_flowrate_CORE : 1100
[WARNING] fgt_get_sensorValue sensor channel 1934189570 of type F_OEM error 14 - Sensor is calibrating Controller SN: 13462 info: remote control
[WARNING] fgt_get_sensorValue sensor channel 1934189570 of type F_OEM error 14 - Sensor is calibrating Controller SN: 13462 info: remote control
falcon SHELL, setting calibration to IPA
OK
max_flowrate_CTU : 600
max_flowrate_SHELL : 120
max_flowrate_CORE : 1100
[WARNING] fgt_get_sensorValue sensor channel 1934189572 of type F_OEM error 14 - Sensor is calibrating Controller SN: 13462 info: remote control
[WARNING] fgt_get_sensorValue sensor channel 1934189572 of type F_OEM error 14 - Sensor is calibrating Controller SN: 13462 info: remote control
falcon CORE, setting calibration to HFE
[WARNING] fgt_set_sensorCalibration sensor channel 1934189571 of type F_OEM error 2 - Flowunit does not support calibration table Controller SN: 13462 info: remote control
max_flowrate_CTU : 600
max_flowrate_SHELL : 600
max_flowrate_CORE : 1100
falcon CORE, setting calibration to FC40
[WARNING] fgt_set_sensorCalibration sensor channel 1934189571 of type F_OEM error 2 - Flowunit does not support calibration table Controller SN: 13462 info: remote control
max_flowrate_CTU : 600
max_flowrate_SHELL : 600
max_flowrate_CORE : 1100
falcon CORE, setting calibration to H2O
max_flowrate_CTU : 600
max_flowrate_SHELL : 600
max_flowrate_CORE : 1100`

Any clue what we could do ? 
Best regard ,

Ygor-Oliveira, 2025-08-25:
> The following code produce the following output ( sorry for the lenght of them ) : 

You can attach files instead of the code blocks, or hide the code with a `/Details` block, so it is easier to scroll through the responses.

> also , I had to add a 1 second sleep to get the value getting changed , could it be the workaround ?

Indeed, when you send a command to change the calibration, the sensor has to receive it and then send a response confirming that it has been changed, before the range can be updated. This should take 200ms at most. The `fgt_init` function returns immediately if it doesn't find any new instruments to initialize, so it tries to update the range before the sensor is ready. We've tried to avoid blocking functions while waiting for the instruments because it could block GUI programs, so sometimes you have to wait manually. I'll see to improve this workflow and add this information to the SDK documentation.

But with the delay, does it work correctly? Or is there still another problem? The L range returns an error when setting the calibration to HFE, FC40 or Oil, but that is expected, as only the M range was calibrated for these liquids. Other than that, it seems to return the correct ranges in this example.

JeremieBours0, 2025-08-25:
> so sometimes you have to wait manually.

Ok understood , I will add a 500ms blocking time in the code to wait for sensor being updated. 
That is a good solution for us !

Here are the condensed calibration respond from the sensors , I see that HFE , FC40 , OIL ranges are all = 200. 
Could you comfirm ? I cant find it on the docs sorry.

```
<details>
<summary>Sensor 1</summary>
sensor 1934189570, setting calibration to H2O
sensorRange after calib : 120.0

sensor 1934189570, setting calibration to IPA
sensorRange after calib : 600.0

sensor 1934189570, setting calibration to HFE
sensorRange after calib : 200.0

sensor 1934189570, setting calibration to FC40
sensorRange after calib : 200.0

sensor 1934189570, setting calibration to OIL
sensorRange after calib : 200.0
</details>

```

```
<details>
<summary>Sensor 2</summary>

sensor 1934189572, setting calibration to H2O
sensorRange after calib : 120.0

sensor 1934189572, setting calibration to IPA
sensorRange after calib : 600.0

sensor 1934189572, setting calibration to HFE
sensorRange after calib : 200.0

sensor 1934189572, setting calibration to FC40
sensorRange after calib : 200.0

sensor 1934189572, setting calibration to OIL
sensorRange after calib : 200.0

</details>

```
```
<details>
<summary>Sensor 3</summary>
sensor 1934189571, setting calibration to H2O
sensorRange after calib : 1100.0

sensor 1934189571, setting calibration to IPA
sensorRange after calib : 11000.0
</details>

```

> The L range returns an error when setting the calibration to HFE, FC40 or Oil, but that is expected, as only the M range was calibrated for these liquids. 

Yes , you are right,  I copy past too much , my bad 

If we comfirme  "HFE , FC40 , OIL ranges are all = 200" , 
everything is then fixed , thank you !

Best regard

PR #69 (2023-06-16): Update SDK to version 23.0.0.0
### Added
- Support for custom Switchboard with 16 Two-Switch ports
- The SDK can now log events for debugging
- Function `fgt_set_log_verbosity` to enable and disable event logs
- Function `fgt_set_log_output_mode` to set how event logs are returned to the user
- Function `fgt_get_next_log` to retrieve event logs manually as strings

### Changed
- Switchboard ports A to D are now mapped to port numbers 16 to 19, instead of 8 to 11.

### Fixed
- Fixed spurious errors being returned by `fgt_get_sensorStatus`.
- F-OEM no longer returns brief connection errors when  [... cut]
closed 2025-08-25 by JeremieBours0