Fluigent/fgt-SDK #70: Library crashs under Raspberry Pi (Linux ARM64)

jporcher, 2023-07-12:
Hello,

I'm trying to connect a Fluigent controller from a Raspberry Pi.
Using SDK version 23.0.0. I wrote a very simple sample program:

```
#include "fgt_SDK.h"
#include <iostream>

int main( int argc, char* argv[] )
{
    std::cout << "Calling fgt_init" << std::endl;
    fgt_init();
    std::cout << "Calling fgt_init" << std::endl;
    return 0;
}
```

Then, I compile, link to libfgt_SDK.so and run this program. If no Fluigent device is connected to the Rapsberry, it just works fine and return 0 (this confirms libfgt_SDK.so is correctly loaded and usable).
But, when a Fluigent device is connected, the program crashs when `fgt_init()` is called. I compiled in debug and ran from GDB, got the output:

```
Thread 1 "fluigent_bin" received signal SIGABORT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50          ../sysdeps/unix/sysv/linux/raise.c: Aucun fichier ou dossier de ce type.
```

Trace is:
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x0000007ff7d94aa0 in __GI_abort () at abort.c:79
#2  0x0000007ff7a70b24 in _Unwind_SetGR () from /lib/aarch64-linux-gnu/libgcc_s.so.1
#3  0x0000007ff7a75464 in __gcc_personality_v0 () from /lib/aarch64-linux-gnu/libgcc_s.so.1
#4  0x000000555564a16c in _Unwind_RaiseException_Phase2 ()
#5  0x000000555564a6b8 in _Unwind_RaiseException ()
#6  0x00000055556351f8 in __cxa_throw ()
#7  0x0000007ff7f220ac in ?? ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#8  0x0000007ff7f4f508 in ?? ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#9  0x0000007ff7f688e8 in ?? ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#10 0x0000007ff7f1cc68 in ?? ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#11 0x0000007ff7f4f75c in ?? ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#12 0x0000007ff7f53574 in ?? ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#13 0x0000007ff7f6885c in ?? ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#14 0x0000007ff7f532bc in ?? ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#15 0x0000007ff7f5350c in ?? ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#16 0x0000007ff7f6ac5c in ?? ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#17 0x0000007ff7f68030 in ?? ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#18 0x0000007ff7f794e0 in ?? ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#19 0x0000007ff7f912e4 in fgt_init ()
   from /home/user/dev/vobs_uflu/build/fluigent2/linarm64/debug/stg/Debug/libfgt_SDK.so
#20 0x00000055555b5d00 in main (argc=1, argv=0x7fffffef58)
    at /home/user/dev/vobs_uflu/private/uflu/prg/fluigent/src/main.cpp:8
```

If you cannot reproduce and send me a debug version of libfgt_SDK.so, I may be able to send you a more verbose trace.

I'm using:
Linux raspberrypi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
With G++:
gcc version 10.2.1 20210110 (Debian 10.2.1-6)

Ygor-Oliveira, 2023-07-12:
Hello,

We could not reproduce the issue on our ARM64 system, which has the following configuration:

Linux retd-desktop 5.11.0-1027-raspi #30-Ubuntu SMP PREEMPT Thu Jan 13 12:52:48 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
g++-8 (Ubuntu/Linaro 8.4.0-7ubuntu3) 8.4.0
Instruments: LineUP Link + Flow EZ

Unfortunately, we will not be able to build a debug library at the moment. We are going to inspect the code to see if there is any condition that could cause an unhandled exception. In the meantime, here are some things you can do to help us pinpoint the problem:

- Let us know which types of Fluigent instruments you are using.
- Make sure you have run the linux-udev.sh script to give the correct permissions to the USB devices.
- Run one of the [C++ examples](https://github.com/Fluigent/fgt-SDK/tree/23.0.0/C%2B%2B), such as Basic Get Instruments Info, using the provided CMake project, to make sure it is not a compiler configuration problem.
- Add the call `fgt_set_log_verbosity(5)` before `fgt_init()`, to see if it is able to log any data before it crashes. If it does, please send us the log file it creates.
- Try the previous release, [22.2.0.0](https://github.com/Fluigent/fgt-SDK/releases/tag/22.2.0.0), since the latest one has the logging feature, which introduces some additional complexity and requires more recent versions of libc and libc++, so it might be more likely to cause compatibility issues.

Thank you in advance and sorry for the inconvenience.

Best regards,
Ygor

jporcher, 2023-07-12:
OK, I'll try all this next week and come back to you.

If it helps, note that fgt_detect crashs the same way.

Kind regards,
Jean

closed 2023-07-17 by Ygor-Oliveira