Fluigent/fgt-SDK #72: Pluging Fluigent device with other USB devices messes up udev?

jporcher, 2023-07-21:
I'm experiencing problems when connecting both a Fluigent device and a FTDI device to a Raspberry Pi.

I ran linux-udev.sh as root. Now when being root everything seems to work fine, but when not being root the USB enumeration becomes unstable.

See this stackexchange post for more details:
https://unix.stackexchange.com/q/752054/202228

Ygor-Oliveira, 2023-07-21:
Hello,

Do you see the issue only when you have both the MFCS and the FTDI device together, but not with one or the other? Do you know the VID and PID of your device? I was not able to reproduce the issue with our Raspberry Pi using only a Fluigent instrument, and I don't have another device to test right now. We are going to look into it in more detail next week. The rules file created by the script is only supposed to give read and write permissions to Fluigent instruments. It should not change anything else. You can run `ls -l /dev/hidraw*` with an instrument connected to see if the permissions were set correctly. It should be `cwr-rw-rw-` for the device corresponding to the instrument.

Best regards,
Ygor

jporcher, 2023-08-16:
Hello,

Sorry for the late reply, I was on vacation.

I was experiencing the problem when both MFCS and FTDI devices are plugged. The FTDI device's VID is 0x0403 and PID 0x6001.

Now, I cannot reproduce the problem anymore, not so easily at least. I could see it again when plugging 3 MFCS + 2 FTDI devices with 2 hubs in a daisy chain. Plugging each hub to different USB ports works, when first hub is connected to the second hub, it starts failing. It may be a hub or a power issue, but it's wierd using "sudo" hides the problem then.
I can't remember if I was using Raspberry USB ports differently in the past (for mouse and keyboard I mean) when I was able to reproduce the issue more easily.

I tested hidraw privileges and it's as you say it should be.

jporcher, 2023-08-18:
Hello Ygor,

Thank you for your help. Feel free to close this Issue, I'll reopen if I have more info showing that the problem comes from your SDK (which is most likelly not the case, I bet the hub is the problem as you mentioned).

Kind regards,
Jean

closed 2023-08-22 by mtrellet