Fluigent/fgt-SDK #73: Plexflo96 flow control issue

jmschreiner, 2023-08-21:
Hi, I am working on the software for the Plexflo96 for Bruker, and we're having an intermittent issue with the flow control.  We have two different pressure controllers, which push the flow in opposite directions through the flow sensor when given positive pressure.  To account for this, we change the pressure scale to account for the the change in direction when using the two pressure sources.  I filed an issue a few months ago related to this, and that solution mostly seems to work fine.  However, once in a while, it acts as if the sign is still inverted, resulting in the pressure continually increasing and the flow increasing beyond what was requesting.  I copied a log segment below showing the exact same sequencing of fgt calls, where one results in uncontrolled flow, and the other results in properly controlled flow.  I am wondering if there is a timing issue, or some other known issue in the sdk?  As far as I know, the flow is taking the correct path through the valve network - it is just not regulated correctly.  We are currently using dll version 22.2.0.0.  Please help me get to the bottom of this. 

Thanks,
John

[2023-07-26T12:27:04.362] starting fluidics micro step  2 :
[2023-07-26T12:27:04.367]   name:  "Sequence 4"
[2023-07-26T12:27:04.371]   typeStr: "prime injection"
[2023-07-26T12:27:04.375]   sequence: 3
[2023-07-26T12:27:04.379]   subSequence: 1
[2023-07-26T12:27:04.383]   buffer name: "Buffer 1"
[2023-07-26T12:27:04.387]   source: "Reservoir *TBD*"
[2023-07-26T12:27:04.391]   target: "Waste"
[2023-07-26T12:27:04.395]   waitForBufferLoad: true
[2023-07-26T12:27:04.399]   pumpSpeed: 500
[2023-07-26T12:27:04.403]   pumpVolume: 151.31
[2023-07-26T12:27:04.407]   pumpTime: 0
[2023-07-26T12:27:04.411]   pumpContinuous: false
[2023-07-26T12:27:04.415]   onlyIfNotFlushed: false
[2023-07-26T12:27:04.419]   pauseTime: 0
[2023-07-26T12:27:04.423]   bleach: false
[2023-07-26T12:27:04.427]   pressureThreshold: 0
[2023-07-26T12:27:04.430]   pressureThresholdTimeout: 0
[2023-07-26T12:27:04.434]   releasePressure: true
[2023-07-26T12:27:04.438]   markClean:  false
[2023-07-26T12:27:04.442]   markDirty:  true
[2023-07-26T12:27:04.446]   markFlushed:  false
[2023-07-26T12:27:04.450]   flushedBuffer:  ""
[2023-07-26T12:27:04.453] Fluidics state changed:  "STATE_BUFFER_CHECK"
[2023-07-26T12:27:04.557] Fluidics state changed:  "STATE_MOVE_VALVES"
[2023-07-26T12:27:04.654] setting fluigent source to reservoir:  "R1"
[2023-07-26T12:27:04.658] setting fluigent target to waste
[2023-07-26T12:27:04.663] FluigentFGTController: setting position:  "R1"           // function tracing
[2023-07-26T12:27:04.667] FluigentFGTController: setting valves:  -1 -1 2          // function tracing
[2023-07-26T12:27:04.671] FluigentFGTController: setting output:  waste            // function tracing
[2023-07-26T12:27:04.675] FluigentFGTController: setting chip index:  "n/a"        // function tracing
[2023-07-26T12:27:04.679] FluigentFGTController: setting ppu:  output              // function tracing
[2023-07-26T12:27:04.683] FluigentFGTController: setting output to  1              // Fgt_set_valvePosition(output 2 switch, 1, fgt_SWITCH_DIRECTION::Shortest, 1);
[2023-07-26T12:27:04.839] FluigentFGTController: setting PPU to  0                 // Fgt_set_valvePosition(ppu 2 switch, 0, fgt_SWITCH_DIRECTION::Shortest, 1);
[2023-07-26T12:27:04.996] Fluidics state changed:  "STATE_MOVE_VALVES_WAIT"
[2023-07-26T12:27:05.439] Fluidics state changed:  "STATE_PUMP"
[2023-07-26T12:27:05.543] FluigentFGTController: setting flow rate to  500 uL/min  // function tracing
[2023-07-26T12:27:05.548] setting flow sensor scale: 1                             // Fgt_set_sensorCustomScale(flow sensor, 1, 0, 0)
[2023-07-26T12:27:05.552] setting flow regulation: 1 500                           // Fgt_set_sensorRegulation(flow sensor, 1, 500)
[2023-07-26T12:27:05.556] Fluidics state changed:  "STATE_PUMP_PRESSURIZE"
[2023-07-26T12:27:05.903] fluidics skipping prepressurization
[2023-07-26T12:27:05.908] FluigentFGTController: setting flow enabled:  true       // function tracing
[2023-07-26T12:27:05.912] setting valve position:  2                               // Fgt_set_valvePosition(m switch, 2, fgt_SWITCH_DIRECTION::Shortest, 1)
[2023-07-26T12:27:06.825] Fluidics state changed:  "STATE_PUMP_WAIT"
[2023-07-26T12:27:09.546] fluidics current pressure:  403.64  flow:  686.345       // uncontrolled flow rate
[2023-07-26T12:27:14.531] fluidics current pressure:  462.656  flow:  650.759      // uncontrolled flow rate
[2023-07-26T12:27:19.518] fluidics current pressure:  486.009  flow:  819.862      // uncontrolled flow rate
[2023-07-26T12:27:19.644] Fluidics state changed:  "STATE_PAUSE"
[2023-07-26T12:27:20.019] FluigentFGTController: setting flow rate to  0 uL/min    // Fgt_set_pressure(pressure channel, 0); Fgt_set_pressure(push/pull channel, 0);
[2023-07-26T12:27:20.024] FluigentFGTController: releasing pressure                // Fgt_set_pressure(pressure channel, 0); Fgt_set_pressure(push/pull channel, 0);
[2023-07-26T12:27:20.034] FluigentFGTController: setting flow enabled:  false      // function tracing
[2023-07-26T12:27:20.038] setting valve position:  0                               // Fgt_set_valvePosition(m switch, 0, fgt_SWITCH_DIRECTION::Shortest, 1)
[2023-07-26T12:27:20.953] Fluidics state changed:  "STATE_FINISH"

[2023-07-26T12:29:07.888] starting fluidics micro step  5 :
[2023-07-26T12:29:07.893]   name:  "Sequence 4"
[2023-07-26T12:29:07.897]   typeStr: "prime injection"
[2023-07-26T12:29:07.900]   sequence: 3
[2023-07-26T12:29:07.904]   subSequence: 1
[2023-07-26T12:29:07.908]   buffer name: "Buffer 1"
[2023-07-26T12:29:07.912]   source: "Reservoir *TBD*"
[2023-07-26T12:29:07.916]   target: "Waste"
[2023-07-26T12:29:07.920]   waitForBufferLoad: true
[2023-07-26T12:29:07.924]   pumpSpeed: 500
[2023-07-26T12:29:07.928]   pumpVolume: 151.31
[2023-07-26T12:29:07.932]   pumpTime: 0
[2023-07-26T12:29:07.936]   pumpContinuous: false
[2023-07-26T12:29:07.939]   onlyIfNotFlushed: false
[2023-07-26T12:29:07.943]   pauseTime: 0
[2023-07-26T12:29:07.947]   bleach: false
[2023-07-26T12:29:07.951]   pressureThreshold: 0
[2023-07-26T12:29:07.955]   pressureThresholdTimeout: 0
[2023-07-26T12:29:07.959]   releasePressure: true
[2023-07-26T12:29:07.963]   markClean:  false
[2023-07-26T12:29:07.967]   markDirty:  true
[2023-07-26T12:29:07.970]   markFlushed:  false
[2023-07-26T12:29:07.974]   flushedBuffer:  ""
[2023-07-26T12:29:07.978] Fluidics state changed:  "STATE_BUFFER_CHECK"
[2023-07-26T12:29:08.074] Fluidics state changed:  "STATE_MOVE_VALVES"
[2023-07-26T12:29:08.221] setting fluigent source to reservoir:  "R1"
[2023-07-26T12:29:08.226] setting fluigent target to waste
[2023-07-26T12:29:08.231] FluigentFGTController: setting position:  "R1"
[2023-07-26T12:29:08.235] FluigentFGTController: setting valves:  -1 -1 2
[2023-07-26T12:29:08.238] FluigentFGTController: setting output:  waste
[2023-07-26T12:29:08.242] FluigentFGTController: setting chip index:  "n/a"
[2023-07-26T12:29:08.246] FluigentFGTController: setting ppu:  output
[2023-07-26T12:29:08.250] FluigentFGTController: setting output to  1
[2023-07-26T12:29:08.406] FluigentFGTController: setting PPU to  0
[2023-07-26T12:29:08.563] Fluidics state changed:  "STATE_MOVE_VALVES_WAIT"
[2023-07-26T12:29:08.830] Fluidics state changed:  "STATE_PUMP"
[2023-07-26T12:29:09.012] FluigentFGTController: setting flow rate to  500 uL/min
[2023-07-26T12:29:09.017] setting flow sensor scale: 1
[2023-07-26T12:29:09.021] setting flow regulation: 1 500
[2023-07-26T12:29:09.025] Fluidics state changed:  "STATE_PUMP_PRESSURIZE"
[2023-07-26T12:29:09.120] fluidics skipping prepressurization
[2023-07-26T12:29:09.125] FluigentFGTController: setting flow enabled:  true
[2023-07-26T12:29:09.129] setting valve position:  2
[2023-07-26T12:29:10.042] Fluidics state changed:  "STATE_PUMP_WAIT"
[2023-07-26T12:29:12.984] fluidics current pressure:  325.374  flow:  497.103    // properly controlled flow rate
[2023-07-26T12:29:18.101] fluidics current pressure:  333.579  flow:  497.655    // properly controlled flow rate
[2023-07-26T12:29:23.007] fluidics current pressure:  333.895  flow:  498.759    // properly controlled flow rate
[2023-07-26T12:29:28.060] fluidics current pressure:  333.579  flow:  495.724    // properly controlled flow rate
[2023-07-26T12:29:28.593] Fluidics state changed:  "STATE_PAUSE"
[2023-07-26T12:29:28.912] FluigentFGTController: setting flow rate to  0 uL/min
[2023-07-26T12:29:28.917] FluigentFGTController: releasing pressure
[2023-07-26T12:29:28.929] FluigentFGTController: setting flow enabled:  false
[2023-07-26T12:29:28.933] setting valve position:  0
[2023-07-26T12:29:29.846] Fluidics state changed:  "STATE_FINISH"

jmschreiner, 2023-08-22:
Thanks Ygor, I'll try it out.
J

jmschreiner, 2023-08-22:
I have a question about how to use the fgt logging.  I would like to incorporate the fgt log messages into our own log.  Are the fgt log messages only generated directly in the api functions themselves, or are there threads run by the fgt api in the background that may also be generating messages?  Is it ok to just check for messages after every fgt api call, or do we need to periodically check for messages even if our fluidics system is not actively being used and we're not making any calls?

Ygor-Oliveira, 2023-08-23:
The internal threads can log events as well. For example, it logs a warning when an instrument is disconnected, or when the sensor regulation is stopped automatically. If you use the maximum verbosity (5), it will also log the data exchanged with the instruments every few milliseconds. The internal queue can hold 1 million entries before it starts deleting the old ones, so in theory you only need to flush it once every few hours, but in this case the logs will be out of order in your file, and you will have to sort them by timestamp afterward. In our apps, we flush the SDK log queue every time the app itself wants to log an event, so the order is preserved.

jmschreiner, 2023-09-29:
Hi Ygor, we were able to capture an instance of the flow regulation not working correctly with verbosity 5.  The fluigent log messages are relayed into our own application log, which is attached.  

If for example, you look at 2023-09-29T14:36:34.211 in the log, we are setting the flow regulation to 500, but the reported pressure and flow just keep climbing until they reach 681 mbar and 1056 ul/min respectively at 2023-09-29T14:37:08.142, before we stop the flow.  

Our user reports that everything was working fine for a whole day until something happened, and after that all flow was unregulated.  

Can you find anything useful in the fgt log messages that will help figure out what is happening?

[fluigent log.zip](https://github.com/Fluigent/fgt-SDK/files/12772065/fluigent.log.zip)

closed 2024-04-04 by mtrellet