Fluigent/fgt-SDK #74: support on macOS running with Apple Silicon CPU

time4Wiley, 2023-08-29:
There is support for x64 for macOS. Can we make an arm64 version for macOS?

time4Wiley, 2023-08-29:
`    if not(os.path.exists(_libpath)): _raise_system_not_supported()
                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/wei/ai-wiley/nx/fgt-SDK/Python/Fluigent/SDK/low_level.py", line 28, in _raise_system_not_supported
    raise NotImplementedError("SDK not supported on {} with architecture {} ({} bits)".
NotImplementedError: SDK not supported on darwin with architecture arm64 (64 bits)`

time4Wiley, 2023-08-31:
the SDK works OK with Python installed using pyenv inside Terminal run by Rosetta. Would be great if have native support on ARM64

mtrellet, 2023-08-31:
Hello,

Thanks for your feedback and sorry for our answer delay. 
As of today, we have indeed no native support for Mac with ARM chips. We have listed this new support in our next development tasks and put it with high priority. We cannot guarantee this support anytime soon but we'll definitely do our best to add this before next version of the SDK. We hope that the workaround using Rosetta is fine for you for the time being.

Best,
Mikael

closed 2025-01-17 by mtrellet