Fluigent/fgt-SDK #75: Oscillating behaviour in the response of my Fluigent LineUp™ series System when using command fgt_set_sensorRegulation()

kithminrw, 2023-10-06:
Oxygen version: 1.1.0.0
OS: Windows 10 Pro
Device and Firmware Information: [Fluigent.txt](https://github.com/Fluigent/fgt-SDK/files/12834169/Fluigent.txt)

Program command series: 
1. Initiate Fluigent
2. Use a python script with the fgt-SDK
3. fgt_init() and follow the BasicSensorRegulation example
4. Set any flow rate using fgt_set_sensorRegulation() command

Results:
[image]

[image]

[image]

I need to understand where this oscillatory behavior is coming from. Is it due to the controller?

Ygor-Oliveira, 2023-10-09:
Hello,

The system can become difficult to control if the fluidic resistance is low compared to the range of your pressure controller, or if there is a significant physical delay between the pressure and the flow rate (e.g., due to long and/or flexible tubing). In these cases, you can use the [fgt_set_sensorRegulationResponse](https://raw.githubusercontent.com/Fluigent/fgt-SDK/23.0.0/Fluigent%20SDK.pdf#subsection.5.2.39) function to increase the response time and force the regulation to take smaller steps, which should mitigate the oscillations.

Best regards,
Ygor

mtrellet, 2024-04-04:
Hello @kithminrw,

Did Ygor's reply help for your problem? Can I close this issue or do you have other questions?

closed 2024-04-12 by mtrellet