Fluigent/fgt-SDK #77: Flow Unit M+ disconnect from FlowEZ Push/Pull leading to pressure spike

ljpietropaoli-mekonos, 2023-11-21:
The Push/Pull unit will sometimes lose connection with the Flow Unit. When this happens, it seems to set the flow rate to 0 in the control loop and thus spikes the pressure to maximum (see pressure graph, where the large negative spike is associated with the first of the error messages you can see in the log). We'd like to prevent the unit from losing connection and, barring that, would like to change the error handling so that it pauses the control loop until connection reestablishes instead of spiking the pressure. Any suggestions? Would this require a firmware update or is it some API setting that we missed? 

[image]

Ygor-Oliveira, 2023-11-22:
Hello,

Could you rerun your experiment setting `fgt_set_log_verbosity(5)` at the start (before any other SDK function), and send us the file it generates? That might help us determine why your device is behaving this way. The log file size is reduced significantly by compressing it to a ZIP archive. You can attach it here or send it to support@fluigent.com. Thank you in advance.

Best regards,
Ygor

ljpietropaoli-mekonos, 2023-11-22:
Hi Ygor, 

Thanks for the response. I will add the suggested function call and attempt to recreate this issue. 

Just to verify, is there no way for me to adjust the PID parameters on the Push/Pull device? Or to add behavior such that if the sensor disconnects the pump does not register a reading of "0" and continue to drive the pressure higher?

mtrellet, 2024-04-04:
Hello @ljpietropaoli-mekonos ,

Did Ygor's reply help for your problem? Can I close this issue or do you have other questions?

ljpietropaoli-mekonos, 2024-04-05:
Yes, I have found a workaround for now and hope to see the change in the next release. Please feel free to close, thank you!

mtrellet, 2024-04-08:
Thanks for your feedback! This is indeed planned for next release.

closed 2024-04-08 by mtrellet