Fluigent/fgt-SDK #79: Using fgt-SDK with Xcode on a Mac

liubiofluids, 2023-12-29:
Is it possible to program in Xcode or g++? I always get linker errors like the following: 

ld: warning: dylib (.//libfgt_SDK.dylib) was built for newer macOS version (11.2) than being linked (11.0)
Undefined symbols for architecture x86_64:
  "Fgt_get_pressure(unsigned int, float*)", referenced from:
      _main in Basic Set Pressure-c258ae.o
  "Fgt_set_pressure(unsigned int, float)", referenced from:
      _main in Basic Set Pressure-c258ae.o
  "Fgt_get_pressureRange(unsigned int, float*, float*)", referenced from:
      _main in Basic Set Pressure-c258ae.o
  "Fgt_init()", referenced from:
      _main in Basic Set Pressure-c258ae.o
  "Fgt_close()", referenced from:
      _main in Basic Set Pressure-c258ae.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Ygor-Oliveira, 2024-01-03:
Hello,

Did you build using the commands provided in the example [readme](https://github.com/Fluigent/fgt-SDK/blob/23.0.0/C%2B%2B/README.md)? Can you paste the full output of the command? It is possible that the linker is failing due to the macOS version. 11.0 is no longer supported by Apple, and we do not have it available to test. However, note that the symbols it is unable to find are defined in the middleware file [fgt_SDK_Cpp.cpp](https://github.com/Fluigent/fgt-SDK/blob/23.0.0/C%2B%2B/fgt_SDK_Cpp/fgt_SDK_Cpp.cpp), so this file must be included in your project. The examples include it automatically when using the provided CMake commands. The readme also has a section explaining how to modify the example to create your own project.

Best regards,
Ygor

mtrellet, 2024-04-04:
Hello @liubiofluids ,

Did Ygor's reply help for your problem? Can I close this issue or do you have other questions?

liubiofluids, 2024-04-05:
We figured it out. It is fine to close the issue.

mtrellet, 2024-04-08:
Thanks for your reply, I close the issue

closed 2024-04-08 by mtrellet