Fluigent/fgt-SDK #80: FlowEZ firmware versions 5.51 and 5.52 have issues with fgt-SDK / LabView Integration. Return "NaN" values where previous versions did not.

ametbb, 2024-02-09:
FlowEZ firmware versions 5.51 and 5.52 have issues with fgt-SDK / LabView Integration. Return "NaN" values where previous versions did not.

Ygor-Oliveira, 2024-02-12:
Hello,

Could you send us a log? It would help us identify the problem. You have to call the `fgt set log verbosity` VI at the start of your program (before any other calls to the fgt_SDK), setting the verbosity input to 5, and then build your project as a desktop application and run it. It will produce a .log file in the same directory as the application executable, which you can attach here or send to <support@fluigent.com>. You can compress it to .zip to reduce the size significantly if it does not fit in an attachment.

Thank you in advance.

Best regards,
Ygor

ametbb, 2024-02-13:
- The problem appears to have been solved by upgrading the Fluigent SDK (Labview) to the latest version 23.0.0
- The problematic combination was FlowEZ controllers with firmware 5.51 and 5.52 and labview written using older Fluigent SDK versions, though FlowEZ controllers with firmware V4.02 & V3.02 were fine withe previous Fluigent SDKs.

closed 2024-02-13 by ametbb