Fluigent/fgt-SDK #81: How do I programmatically get the pressure index given the sensor index and vice versa?

faruqsandi, 2024-02-20:
Hello there!

I want to use fgt_set_sensorRegulation to control the flow rate. Both the sensor index and the pressure index are required parameters. How do I identify the pair if I only know one of them?

I assume that is possible because in Oxygen, I can enable regulation of Flow EZ or Flow Unit while disabling another automatically
[image]

Thanks in advance!

Ygor-Oliveira, 2024-02-20:
Hello,

There is no function to retrieve the channel indices of a running regulation. You have to keep track of the indices you gave to `fgt_set_sensorRegulation`. To find which index corresponds to which channel, you can use `fgt_get_pressureChannelsInfo` and `fgt_get_sensorChannelsInfo`, which return the index alongside other information like the serial number and the position/port number. OxyGEN uses this information to automatically associate a Flow Unit with the Flow EZ it is physically connected to, for convenience, but you can choose a different channel in the drop-down menu, whose options are derived from the output of `fgt_get_pressureChannelsInfo`. When using the SDK directly, there is no such automatic association, so you have to go through the lists to find the channels you want to use. Note that the indices of the channels will not change unless you call `fgt_close` or `fgt_initEx`, so generally you only need to retrieve them once.

Best regards,
Ygor

faruqsandi, 2024-02-23:
Hello, Ygor.

Here, I've included the output from both channels. How do I know that, say, pressure index 0 is physically connected to sensor index 0?
Is it done by comparing the DeviceSN of a pressure channel and a sensor channel?

Thank you.

#### fgt_get_sensorChannelsInfo()
```
>>> fgt_get_sensorChannelsInfo()
([

<class 'Fluigent.SDK.fgt_CHANNEL_INFO'>
ControllerSN: 12776
firmware: 5.52
DeviceSN: 15630
position: 0
index: 0
indexID: 1126032384
InstrType: LineUP,

<class 'Fluigent.SDK.fgt_CHANNEL_INFO'>
ControllerSN: 12776
firmware: 5.52
DeviceSN: 15655
position: 1
index: 1
indexID: 1126032385
InstrType: LineUP], [fgt_SENSOR_TYPE: 5 (Flow_M_dual), fgt_SENSOR_TYPE: 5 (Flow_M_dual)

])
```
#### fgt_get_pressureChannelsInfo()
```
>>> fgt_get_pressureChannelsInfo()
[
<class 'Fluigent.SDK.fgt_CHANNEL_INFO'>
ControllerSN: 12776
firmware: 5.52
DeviceSN: 15630
position: 0
index: 0
indexID: 1126031872
InstrType: LineUP, 

class 'Fluigent.SDK.fgt_CHANNEL_INFO'>
ControllerSN: 12776
firmware: 5.52
DeviceSN: 15655
position: 1
index: 1
indexID: 1126031873
InstrType: LineUP

]
>>> 
```

Ygor-Oliveira, 2024-02-23:
Hello,

The most complete method is to compare the `InstrType`, `ControllerSN`, and `position`. `ControllerSN` is the Link's serial number, and `position` is the Flow EZ's position along the chain. These 3 values should match for the Flow EZ and the Flow Unit connected to it. Comparing the `InstrType` only makes a difference if you happen to have two instruments of different types with the same serial number, which is unlikely with real instruments, but can happen if you are testing with simulated instruments, so it is a good idea to include it to avoid bugs. Comparing the `DeviceSN` would work too, since the Flow Unit does not have a readable serial number, so it returns the Flow EZ's serial number.

faruqsandi, 2024-02-23:
Whoah!  I now get it. Many thanks for that!

closed 2024-02-23 by faruqsandi