Fluigent/fgt-SDK #82: Low effective data refresh rate with flow unit L+ and flowboard

martijnschouten, 2024-02-26:
Hi,

We have been running a flow unit XL in combination with a flowboard for years and it was able to reach an update rate of 50Hz. I now upgraded the XL to an L+ because we reached the maximum flowrate of the XL. However the new sensor seems to update it's flow only once every 1-2 seconds.

Below the output that I get from the flow unit L+ connected to a switchboard when I run the BasicReadSensorData.py example with a 0.1 second sleep for 50 iterations. We are running windows 10, 64 bit. I updated to the newest version of the SDK (version 23.0.0). Is there anything I can do to make it update faster?

Kind regards,

Martijn Schouten
Postdoc EPFL-LMTS

Sensor channel info at index: 0
<class 'Fluigent.SDK.fgt_CHANNEL_INFO'>
ControllerSN: 789
firmware: 0.00
DeviceSN: 0
position: 0
index: 0
indexID: 808539136
InstrType: FRP
Sensor type: Flow_L_plus_dual
Range -40000.00 to 40000.00 µl/min
Measured 62.00 µl/min
Measured -402.00 µl/min
Measured -488.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -520.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -720.00 µl/min
Measured -392.00 µl/min
Measured -466.00 µl/min
Measured -468.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -438.00 µl/min
Measured -160.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -194.00 µl/min
Measured -8.00 µl/min
Measured -2.00 µl/min
Measured 2.00 µl/min
Measured 2.00 µl/min
Measured 2.00 µl/min
Measured 2.00 µl/min
Measured 2.00 µl/min
Measured 2.00 µl/min
Measured 2.00 µl/min
Measured 2.00 µl/min
Measured 2.00 µl/min
Measured 2.00 µl/min
Measured 2.00 µl/min
Measured 2.00 µl/min
Measured 2.00 µl/min

Ygor-Oliveira, 2024-02-26:
Hello,

We were not able to reproduce this issue with the L+ Flow Units we have here. What pressure values did you use to test? In your data, there are a few places where there are multiple consecutive different values. Is it possible that the flow rate variation is simply below the sensor's quantization step most of the time? The step for the L+ is 2 µL/min, whereas for the XL it is 0.2 µL/min.

Could you try with a more dynamic pressure signal like a sine wave? With something like this inside the loop (depending on your physical setup):

```py
import math, time
from Fluigent.SDK import fgt_set_pressure
fgt_set_pressure(0, 50 + 20 * math.sin(time.time() * 0.1))
```

If there is still a problem, could you send us the raw data log so we can analyze it? You can obtain it by adding this at the start of your code:

```py
from Fluigent.SDK import fgt_set_log_verbosity
fgt_set_log_verbosity(5)
```

It will then create a .log file in the folder from which you launched the script.

Best regards,
Ygor

martijnschouten, 2024-02-26:
Hello Igor,

Thank you so much for your fast reply.

We do not have a pressure source that I can control to make a sine in our setup. We use the sensor to measure the flow of our own pumps. In the previous example, I pushed some fluid through the sensor using a syringe. I am sure it was not that constant. When I make a 1 Hz sine save by hand I get the same result. I also tried with a windows 11 X64 PC and that also gives the same result.  

I added the log commands to my code on the Windows 10 PC and ran the code while applying a roughly 1Hz sinusoidal pressure using a syringe. The log file that was created can be found below.
[fgt_sdk_2024-02-26.log](https://github.com/Fluigent/fgt-SDK/files/14406410/fgt_sdk_2024-02-26.log)

Kind regards,

Martijn

Ygor-Oliveira, 2024-02-26:
Hello,

From the logs, the Flowboard is consistently sending data to the PC every 25ms, as it should, but the flow rate measurement remains constant for many data frames. From what I could find, this is an issue with the Flowboard version 1.01 when using M+ and L+ Flow Units. It tries to query them too often and they stop responding. This was fixed in version 1.02. I suggest you write to support@fluigent.com to see if your device can be updated.

Best regards,
Ygor

martijnschouten, 2024-02-27:
Alright, I borrowed a V1.02 flowboard from a colleague and it ran with the proper refresh rate, so that seems to solve that issue. However, after a few times the whole sensor stopped working. It is now not recognised any more at all by both flow boards (fgt_get_sensorChannelsInfo returns []). I tried restarting the computer and unplugging the sensor. The old XL flow sensor does still work. Could this be related to the V1.02 flowboard or is it just bad luck and did we get a bad sensor?

I have attached todays log, it ended up being very long due to all the tries I did to get it to work again.

[fgt_sdk_2024-02-27.log](https://github.com/Fluigent/fgt-SDK/files/14417354/fgt_sdk_2024-02-27.log)

martijnschouten, 2024-04-12:
Yes,  we send our sensor to you to have it checked. Your support team found nothing broken, so they are going to send a new one. I think it breaking probably was just bad luck, so you can close the issue.

closed 2024-04-12 by martijnschouten

mtrellet, 2024-04-12:
Very well, thanks for your quick reply!