Fluigent/fgt-SDK #85: SDK not implemented error

tyquenan, 2024-05-28:
Hi,
I built a PyQt6 application including the fluigent.SDK package, and everything works fine when I run it through my Python IDE.
I tried building and freezing my program into an exe file using pyinstaller, however it crashes on launch with the following error:
>Traceback (most recent call last):
  File "main.pyw", line 7, in <module>
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "PyInstaller\loader\pyimod02_importers.py", line 419, in exec_module
  File "Fluigent\SDK\__init__.py", line 5, in <module>
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "PyInstaller\loader\pyimod02_importers.py", line 419, in exec_module
  File "Fluigent\SDK\low_level.py", line 58, in <module>
  File "Fluigent\SDK\low_level.py", line 28, in _raise_system_not_supported
NotImplementedError: SDK not supported on win32 with architecture AMD64 (64 bits)

From my understanding, AMD64 is the x64 architecture and should be supported by the SDK. I'm not certain whether the error comes from the package or the building process, are such example are scarce.
I'm using Python 3.11 on Windows 11 (AMD64)

tyquenan, 2024-05-28:
For anyone having the same issues, the Fluigent package needs to be manually added to any compilation. It was a quick fix

closed 2024-05-28 by tyquenan

Ygor-Oliveira, 2024-05-29:
Hello,

Our Python package needs a native library to work, and looks for it in a fixed relative path within the package. This does not properly account for executable generation, so it does not work correctly out of the box and returns unhelpful errors for this use case. We are looking for ways to improve this experience. Sorry for the inconvenience.

Best regards,
Ygor

tyquenan, 2024-05-29:
Hopefully the errors are not exclusive to fluigent package, and reproducing fixes for other packages worked. However it did either return this weird error, either returned no error but the program simply didn't launch, so debugging wasn't very straightforward. Hopefully that can be fixed. Thanks!