Fluigent/fgt-SDK #86: Aria to control a vaccum pump issue

Nal44, 2024-05-30:
Hi,

I build as system following the paper : https://www.nature.com/articles/s41596-021-00644-9

I have a leica microscope, and all warner equipment and the aria as in the paper.

the only difference is the chamber since the one in the paper is discontinued... : 
https://www.warneronline.com/open-diamond-bath-imaging-chambers-rc-26-rc-26g

Anyway, I am having issue with evaporation during incubation times, since there is not airflow but the vacuum is on at 37c , then the chamber dries out rapidly.

I made a 3d printed lid which is tight but still it dries out.

hence, I would like to know if we can use an arduino or a controller to control the vacuum pump switch (ON/OFF) then connect it to the ARIA.

So the ARIA can connect with the vacuum controller (via TCP ?) and switch OFF during incubations and back ON once done : Hence using the protocols I could send / signal when needed.

Could it be possible ? that would solve my evaporation problem during incubation time (1h) .

Thanks a lot,
Antho

mtrellet, 2024-05-31:
Hello Antho,

Thanks for the issue description, it is quite complete.
Unfortunately, the pressure pump provided by Fluigent (the [FLPG](https://www.fluigent.com/research/instruments/pressure-sources/flpg-plus/) - as mentioned in the article), does not have any communication capability, either via USB, TTL or anything else.
But I'm not sure this is the "vacuum pump" you are referring to, can you confirm which product exactly you are using?

I think your idea of using a controller is good but would require to find a way to switch ON/OFF the FLPG via this controller, I don't know how this could be done easily. I'll forward this question to our support team via email since this is out of the scope of this SDK. They will contact you as soon as possible.

Best regards,

Mikael

Nal44, 2024-05-31:
Hi Mikael,

Thanks a lot,
Ok in principle that should work, we will be making the setup an order an arduino / make the setup, but I might need help for the TCP client since it is not my expertise (I can code though). Do you have any resources or can you help to build that to ensure the communication between the controller and the ARIA ? 

Thanks a lot,
Antho

mtrellet, 2024-06-03:
Hello Antho,

You should find what you need with a few keywords, "TCP client Arduino example" on Google returns a few hits that are quite helpful ([here](https://forum.arduino.cc/t/link-to-a-very-basic-tutorial-about-tcp-evolved-to-tcp-socket-democode-to-send-receive-data-characters-similar-to-serial/949935/2) and [here](https://arduinogetstarted.com/tutorials/arduino-tcp-client) for instance). Maybe start with some basics about TCP communication and play with Aria UI by sending/receiving commands from/to a simple piece of code written in your favorite language. 
Porting this to Arduino is another step beyond our core expertise and rather specific to your needs so we cannot help there.

Hope this will help,

Best,

Mikael

mtrellet, 2024-09-02:
Hello,

May I close this issue or do you have other questions regarding Fluigent SDK?

Nal44, 2024-09-02:
HI,

Yes, we found a solution using a Wifi plug (shelly) and wrote some code in python to toggle ON/OFF using TCP.
Now we can control the plug via the software, and we fixed the vacuum issue.
We simply turn it OFF before any incubation (wait time ) and ON after for the next step.

Thanks
antho

mtrellet, 2024-09-02:
Very nice to hear! Thanks for your quick answer, good luck with your next experiments then.

closed 2024-09-02 by mtrellet