Fluigent/fgt-SDK #87: Raspberry Pi 4 Compatibility Issues

szapata8, 2024-06-27:
The Fluigent SDK documentation specifies system requirements for different system architectures, and for a Raspberry Pi 4 with ARM64 architecture the following requirements are specified: Requires GLIBC ≥ 2.17, GLIBCXX ≥ 3.4.22, CXXABI ≥ 1.3.11. By running appropriate commands, I see that the Raspberry Pi 4 fulfills all the system requirements (see pictures attached below for versions of each of the requirements). However, when using the python library and calling fgt_set_pressure, an error pops up signifying that there are no modules recognized by the system. The message is the following: "WARNING:root:fgt_set_pressure pressure channel 0 of type NONE error 12 - No modules found".

[image]
[image]
[image]

The Fluigent hardware I'm using works well on a Windows PC, and the information displayed when requesting it from the method fgt_get_pressureChannelsInfo() returns the following information:
[image]

My project has the fgt-SDK/Python/Fluigent directory directly copied from the repository so it does have the libfgt_sdk.so file in the original directory for the corresponding linux/arm64 system architecture.

How can I solve this issue?

mtrellet, 2024-06-28:
Hello,

Thanks for the complete information, very helpful to narrow down the potential problem sources. Before we investigate further though, did you allow access to peripheral devices by running the provided script. As stated in the user manual:
> On Linux, the system usually does not allow access to peripheral devices without superuser rights. It is
possible to make an exception for Fluigent instruments, so the SDK can detect and communicate with them
when run by a normal user. You can run the script linux-udev.sh that we provide, which automatically makes
the necessary changes to the system. The script must be run with superuser rights

This is a first thought, if you did already you should have this file created: `/etc/udev/rules.d/99-fluigent.rules`

szapata8, 2024-06-28:
Thank you, I had actually overlooked running that script. Upon doing so I was able to access the pressure controller modules just fine!

closed 2024-06-28 by szapata8