Fluigent/fgt-SDK #88: Looking for BioRender Graphics

LaceyMesia, 2024-06-27:
Hello Fluigent,

I was pointed to this space to request BioRender graphics that will allow me to recreate the protocol edit screen on the OxyGEN software. I am specifically looking for the tiles. Please let me know if this is something you can send to me.

Lacey Mesia

mtrellet, 2024-06-28:
Hello Lacey,

Sorry but we did not hear about BioRender graphics before your message. 
The protocol editor is made of several custom controls and views made by our team on top of a specific framework. OxyGEN code is unfortunately not open-source...

LaceyMesia, 2024-06-28:
I don't need the code for it, I just need pictographs to make a figure for a paper. Your company gave me something similar for the instrument representations.

mtrellet, 2024-06-28:
I did not understand your point indeed, sorry.
Can you tell us in what format you need those pictographs? Because the tiles of the protocol editor are rendered by the program based on some code and data information processed at runtime. They are not, like the instrument representations, exportable as images or pictures...

mtrellet, 2024-09-02:
May I close this issue or do you have more questions regarding my last comment?

closed 2025-01-17 by mtrellet