Fluigent/fgt-SDK #89: Installing the python API on windows

loop51, 2024-07-03:
Hi,
To install the python sdk I downloaded 
https://github.com/Fluigent/fgt-SDK/releases/download/23.0.0/fgt-SDK-23.0.0.zip
Then used 
python -m pip install fluigent_sdk-23.0.0.zip as per the SDK library

but when I try Python/examples/BasicSetPressure.py (little bit modified, just to test if I can change the pressure set values), I am getting the following error
c:\Users\mdsaifi\Documents\Gamepad_push_pull_interface>python basic_set_pressure_test.py
Traceback (most recent call last):
  File "c:\Users\mdsaifi\Documents\Gamepad_push_pull_interface\basic_set_pressure_test.py", line 5, in <module>
    from Fluigent.SDK import fgt_init, fgt_close
  File "C:\Program Files\Python312\Lib\site-packages\Fluigent\SDK\__init__.py", line 5, in <module>
    from . import low_level
  File "C:\Program Files\Python312\Lib\site-packages\Fluigent\SDK\low_level.py", line 10, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'

I think the sdk install didnt work. 

Thanks

Ygor-Oliveira, 2024-07-03:
Hello,

This was reported before in #78. Your installation is correct, but the method we use to load the native library in the SDK is no longer available by default in Python 3.12. We have fixed that for the next release, but for now, the workaround is to either use an earlier version of Python, or install the setuptools package.

```
python -m pip install setuptools
```

Sorry for the inconvenience.

Best regards,
Ygor

loop51, 2024-07-03:
Thanks. That worked

closed 2024-09-02 by mtrellet