Fluigent/fgt-SDK #94: Delay in Valve Actuation via Python SDK

leonrbt, 2025-01-08:
Hello,

Using Python 3.10, I would like to control a 2-Switch valve connected to my computer via a Switch EZ module and a Link Module. 

I switch the position of the valve using fgt_set_valvePosition from the latest SDK release (23.0.0), which works perfectly fine, but there seems to be a long delay (> 100 ms) which is not there when I control the valves using the Oxygen software. Is this delay hard-coded, and, if so, would there be a way to reduce it or to delete it? I am considering writing serial commands to control the valve directly, but the documentation for this is not fully explicit.

Thank you for your support,

Best regards

mtrellet, 2025-01-09:
Hello,

Theoretically, there should not be any delay when running the `fgt_set_valvePosition` command.
I've just tested, using latest versions of OxyGEN - `v2.3.2` - and the SDK - `v23.0.0` - and I did not notice any delay. For the SDK test I've simply run the sole command to switch the valve, without any operations around.

Do you mind sending us your script to see whether we can reproduce your issue?
And as extra information, how do you estimate the delay?

Thanks in advance

leonrbt, 2025-01-09:
Hello,

Thank you for your answer. 

Below is the script I used:

```
from Fluigent.SDK import fgt_init, fgt_close
from Fluigent.SDK import fgt_get_valveChannelCount, fgt_set_valvePosition
import time

fgt_init()
valve_indices = range(fgt_get_valveChannelCount())
start = time.time()
set_valve_position(valve_indices[0], 0)
delay = time.time() - start
print(delay)
fgt_close()
```

Of course, if the valve was already in position 0, then `delay` is equal to 0.0, which seems to indicate that the delay does not come from the serial communication itself. If the valve was in position 1, then `delay` is equal to 0.18. I also tried to switch back and forth between positions 0 and 1 in a loop of 100 iterations and calling time.time() only before and after the whole loop, which took 36 seconds to complete. The fact that this is consistant with the duration of one switch led me to think there might be a hard-coded delay in the `fgt_set_valvePosition `function.

I did not precisely measure the delay when using OxyGEN (v2.1.0.0), but the switch seems amost instantaneous.

Best

Ygor-Oliveira, 2025-01-09:
Hello,

The `fgt_set_valvePosition` function has a `wait` argument that makes the function block until the instrument reports that the valve has reached the requested position. In Python, this argument is optional and has a default value of `True`, to avoid starting the flow while a valve is still switching when running a simple script. OxyGEN sets it to false to avoid blocking the UI, so this might be the cause of the difference you are observing.

You can call `fgt_set_valvePosition(index, position, 0, 0)` to run the function in non-blocking mode.

closed 2025-01-09 by leonrbt