Fluigent/fgt-SDK #95: Unknown CMake command "COPY_SDK_LIB"

flat4power, 2025-01-12:
Hi,
I use QT Creator 15.0 based on Qt 6.8.1
I try to compile my project with this line on my CmakeLists.txt
target_link_libraries(${PROJECT_NAME} PRIVATE fgt_SDK_cpp)
COPY_SDK_LIB(${PROJECT_NAME})

my folder is like this
/Treefrog_Soft
    /fgt_SDK_Cpp
        /dlls
            fgt_SDK.h
            /windows
               ...
        fgt_SDK_cpp.h
        fgt_SDK_cpp.cpp
   main.cpp
   ...
   
   I need your help because I don't understand where is my problem
   thanks

Ygor-Oliveira, 2025-01-13:
Hello,

We don't have experience with Qt Creator, but it might be simply a matter of adding the SDK's subdirectory to your top-level project file in order to access the function:

```cmake
add_subdirectory(fgt_SDK_Cpp)
```

As `copy_sdk_lib` is not a standard CMake function, but a function we created to copy our native shared library to the output directory:

https://github.com/Fluigent/fgt-SDK/blob/8463d0eb66d935caf5b8d2b9f5c195e2cdedc715/C%2B%2B/fgt_SDK_Cpp/CMakeLists.txt#L57-L61

Let us know if this works for you.

Best regards,
Ygor

flat4power, 2025-01-13:
Thanks for your reply
I made a try with an Example like Basic Set Pressure
the CMakeLists is like this:
`set(program_name "BasicSetPressure")
add_executable(${program_name} "Basic Set Pressure.cpp")
target_link_libraries(${program_name} PRIVATE fgt_SDK_cpp)
COPY_SDK_LIB(${program_name})`
I can compile and execution is ok my controller change his setpoint according to the program

But with the CmakeLists of my project I have an issue :-(

Ygor-Oliveira, 2025-01-13:
Can you show what your entire `CMakeLists.txt` file looks like? Either as an attachment or as a code block.

flat4power, 2025-01-13:
[CMakeLists.txt](https://github.com/user-attachments/files/18396555/CMakeLists.txt)
lot of commented line for different try :-)

Ygor-Oliveira, 2025-01-13:
I tested your file with Visual Studio. I had to remove all Qt related code, but other than that, the only problem I found was that the name of the library was spelled with `Cpp` instead of `cpp`, so it was not linking correctly. After changing that, the build worked. There is an inconsistency between the file and folder name and the library name, which led to this mistake. We will make a note of that to fix it for the next release. So you can try this code to see if it works correctly:

```cmake
add_subdirectory(fgt_SDK_Cpp)
target_link_libraries(${PROJECT_NAME} PRIVATE fgt_SDK_cpp)
COPY_SDK_LIB(${PROJECT_NAME})
```

flat4power, 2025-01-16:
Now it's working
thanks for your help

closed 2025-01-17 by mtrellet